carla-simulator/ros-bridge #533: Sensors synchronization Problem with /carla/objects

Roosstef, 2021-05-03:
Hey all,

I noticed a possible problem with the syncronization between sensor data and /carla/objects.

My setup: 
- Ubuntu 20.04 
- Ros noetic 
- Carla 0.9.11 
- Ros bridge master (4161cf8)

In my example I start the ros_bridge with 2 radar sensors , "sensor.pseudo.tf", "sensor.pseudo.objects" and "sensor.pseudo.markers".  
I use a simple script to print the two topics of the sensors and the /carla/objects 

In the attached file there are different timestamps between the sensor data and the /carla/objects associated with header.seq

[image]

In previous Carla version 0.9.10.1 the same script causes no issues.

joel-mb, 2021-05-31:
Hi @Roosstef. Sorry for the delay.
I think this is definitely a bug in the ROS bridge side. It seems that there is a race condition between the publication of the clock data and the generation of the header for some sensors. I've pushed some changes in https://github.com/carla-simulator/ros-bridge/tree/joel-mb/fix_timestamp that should solve the synchronization problem. Could you verify that it solves the issue for your use case?

Roosstef, 2021-05-31:
@joel-mb your solution fixed the issue. Thank you!

closed 2021-06-15 by joel-mb