carla-simulator/ros-bridge #535: CARLA Waypoint Publisher error

nicema0, 2021-05-05:
roslaunch carla_waypoint_publisher carla_waypoint_publisher.launch
... logging to /home/b/.ros/log/dc41ec26-ad48-11eb-844d-c9454f70dc56/roslaunch-b-MS-7B23-11964.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:34233/

SUMMARY
========

PARAMETERS
 * /carla_waypoint_publisher/host: localhost
 * /carla_waypoint_publisher/port: 2000
 * /carla_waypoint_publisher/role_name: ego_vehicle
 * /carla_waypoint_publisher/timeout: 10
 * /rosdistro: noetic
 * /rosversion: 1.15.9

NODES
  /
    carla_waypoint_publisher (carla_waypoint_publisher/carla_waypoint_publisher.py)

auto-starting new master
process[master]: started with pid [11972]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to dc41ec26-ad48-11eb-844d-c9454f70dc56
process[rosout-1]: started with pid [11982]
started core service [/rosout]
process[carla_waypoint_publisher-2]: started with pid [11989]
2021-05-05 11:23:22.566790: W tensorflow/stream_executor/platform/default/dso_loader.cc:60] Could not load dynamic library 'libcudart.so.11.0'; dlerror: libcudart.so.11.0: cannot open shared object file: No such file or directory; LD_LIBRARY_PATH: /home/b/carla-ros-bridge/catkin_ws/devel/lib:/opt/ros/noetic/lib:/opt/ros/noetic/lib/x86_64-linux-gnu
2021-05-05 11:23:22.566812: I tensorflow/stream_executor/cuda/cudart_stub.cc:29] Ignore above cudart dlerror if you do not have a GPU set up on your machine.
Traceback (most recent call last):
  File "/home/b/carla-ros-bridge/catkin_ws/devel/lib/carla_waypoint_publisher/carla_waypoint_publisher.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/b/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_waypoint_publisher/src/carla_waypoint_publisher/carla_waypoint_publisher.py", line 42, in <module>
    from agents.navigation.global_route_planner import GlobalRoutePlanner
  File "/usr/local/lib/python3.8/dist-packages/agents/__init__.py", line 22, in <module>
    from . import scripts
  File "/usr/local/lib/python3.8/dist-packages/agents/scripts/__init__.py", line 21, in <module>
    from . import train
  File "/usr/local/lib/python3.8/dist-packages/agents/scripts/train.py", line 33, in <module>
    from agents.scripts import configs
  File "/usr/local/lib/python3.8/dist-packages/agents/scripts/configs.py", line 26, in <module>
    from agents.scripts import networks
  File "/usr/local/lib/python3.8/dist-packages/agents/scripts/networks.py", line 30, in <module>
    tfd = tf.contrib.distributions
AttributeError: module 'tensorflow' has no attribute 'contrib'
[carla_waypoint_publisher-2] process has died [pid 11989, exit code 1, cmd /home/b/carla-ros-bridge/catkin_ws/devel/lib/carla_waypoint_publisher/carla_waypoint_publisher.py __name:=carla_waypoint_publisher __log:=/home/b/.ros/log/dc41ec26-ad48-11eb-844d-c9454f70dc56/carla_waypoint_publisher-2.log].
log file: /home/b/.ros/log/dc41ec26-ad48-11eb-844d-c9454f70dc56/carla_waypoint_publisher-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

what should i do?
please help me!

joel-mb, 2021-05-07:
@nicema0 hmm... I think that your problem is that you have two python libraries with the same name `agents` and the `carla_waypoint_publisher` uses the CARLA `agents` module implemented here: https://github.com/carla-simulator/carla/tree/master/PythonAPI/carla/agents. Did you manually add the CARLA egg file to the `PYTHONPATH` or are you using the CARLA debian installation?
Anyway, if you are not using the other `agents` module along with CARLA, I would suggest removing it from the system installation and install it in a virtual environments.

SirMichcik, 2024-10-07:
> @nicema0 hmm... I think that your problem is that you have two python libraries with the same name `agents` and the `carla_waypoint_publisher` uses the CARLA `agents` module implemented here: [...]

Hey, i have the same problem here. AttributeError: module 'tensorflow' has no attribute 'contrib' while trying to use carla-ros-bridge waypoint package. I am using carla full linux build with UE4 editor, so i have installed carla by cloning github repo. Can you explain if and how should i take care of egg files? I didnt add them anywhere during instalation and during my usage of simulator but now they might be the problem. Thank you!