carla-simulator/ros-bridge #541: ModuleNotFoundError: No module named 'ros_compatibility'

pantelmen, 2021-05-07:
Hi , 
i am trying 
`roslaunch carla_ros_bridge carla_ros_bridge.launch`
but i take an error,
any idea for module 'ros_compatibility'

```
> 
> File "/home/pantel/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py", line 13, in <module>
>     from ros_compatibility import (
> ModuleNotFoundError: No module named 'ros_compatibility'
> **================================================================================REQUIRED process [carla_ros_bridge-2] has died!
> process has died [pid 350298, exit code 1, cmd /home/pantel/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py __name:=carla_ros_bridge __log:=/home/pantel/.ros/log/cbbdb370-af5b-11eb-b9a7-8bedc89683a7/carla_ros_bridge-2.log].
> log file: /home/pantel/.ros/log/cbbdb370-af5b-11eb-b9a7-8bedc89683a7/carla_ros_bridge-2*.log
> Initiating shutdown!
> ================================================================================**
> [carla_ros_bridge-2] killing on exit
> [rosout-1] killing on exit
> [master] killing on exit
> shutting down processing monitor...
> ... shutting down processing monitor complete
> done
```

closed 2021-05-08 by pantelmen

August2018, 2021-09-29:
wer

August2018, 2021-09-29:
I get the same error.

cinakun, 2022-04-01:
I get the same error too.