carla-simulator/ros-bridge #555: Problems with : ROS bridge installation for ROS 2 on Docker

egamor18, 2021-06-24:
I trying to install ros bridge for ROS 2 on docker. I am keep getting the error below:

`Starting >>> rviz_carla_plugin                                                                                                                          
Starting >>> carla_walker_agent
--- stderr: rviz_carla_plugin                                                                                                                                      
/root/carla-ros-bridge/src/ros-bridge/rviz_carla_plugin/src/carla_control_panel_ROS2.cpp:25:10: fatal error: tf2/utils.h: No such file or directory
   25 | #include <tf2/utils.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/rviz_carla_plugin.dir/build.make:110: CMakeFiles/rviz_carla_plugin.dir/src/carla_control_panel_ROS2.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /root/carla-ros-bridge/build/rviz_carla_plugin/rviz_carla_plugin_autogen/UVLADIE3JM/moc_carla_control_panel_ROS2.cpp:9,
                 from /root/carla-ros-bridge/build/rviz_carla_plugin/rviz_carla_plugin_autogen/mocs_compilation.cpp:2:
/root/carla-ros-bridge/build/rviz_carla_plugin/rviz_carla_plugin_autogen/UVLADIE3JM/../../../../src/ros-bridge/rviz_carla_plugin/src/carla_control_panel_ROS2.h:17:10: fatal error: rviz_common/panel.hpp: No such file or directory
   17 | #include <rviz_common/panel.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/rviz_carla_plugin.dir/build.make:71: CMakeFiles/rviz_carla_plugin.dir/rviz_carla_plugin_autogen/mocs_compilation.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:79: CMakeFiles/rviz_carla_plugin.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rviz_carla_plugin [5.25s, exited with code 2]
Aborted  <<< carla_ackermann_msgs [5.50s]                                                                                                             
Aborted  <<< carla_manual_control [6.25s]                                                                                             
Aborted  <<< carla_ros_scenario_runner [6.09s]
Aborted  <<< carla_ad_agent [6.14s]
Aborted  <<< carla_walker_agent [6.02s]
Aborted  <<< carla_spawn_objects [6.29s]                        

Summary: 8 packages finished [20.2s]
  1 package failed: rviz_carla_plugin
  6 packages aborted: carla_ackermann_msgs carla_ad_agent carla_manual_control carla_ros_scenario_runner carla_spawn_objects carla_walker_agent
  1 package had stderr output: rviz_carla_plugin
  4 packages not processed
`
Truly, I cant find tf2/utils.h.
What can I do to resolve this rviz_carla_plugin error?
Thank you.

egamor18, 2022-01-25:
> @egamor18 did you resolve this issue? I'm trying to build the ROS2 bridge and fail, but on a completely different step (carla_msgs are "missing"). 

No, its not resolved.

egamor18, 2022-01-25:
> @egamor18 did you resolve this issue? I'm trying to build the ROS2 bridge and fail, but on a completely different step (carla_msgs are "missing").

No, It was not resolved.

nahsimai, 2022-02-11:
> @egamor18 did you resolve this issue? I'm trying to build the ROS2 bridge and fail, but on a completely different step (carla_msgs are "missing").

@MTDzi , did you manage to solve the issues with carla_msgs ?

MTDzi, 2022-02-11:
@nahsimai Yepp, it was a matter of calling:
```
git submodule update --init --recursive
```