carla-simulator/ros-bridge #559: Sensor spawn at wrong position / in wrong tf

Roosstef, 2021-07-13:
Hey,
i noticed that in the latest commit the sensors are spawned in a wrong tf. 
See pictures. Same sensor definition file in different commits.
Top: Commit c9f33e0 Bottom: 9a9eb9b (current master head)
[image]
[image]

Roosstef, 2021-09-13:
Any updates on this issue? 
It seems that this problem still occurs in the latest master.

lasuomela, 2021-09-16:
Seems to me there are multiple issues with the sensors and tf2.

In objects.json, I have defined an rgb camera as follows:

``{``
``                    "type": "sensor.camera.rgb",``
``                    "id": "rgb_front",``
``                    "spawn_point": {"x": 2.0, "y": 0.0, "z": 2.0, "roll": 0.0, "pitch": 0.0, "yaw": 270.0},``
``                    "image_size_x": 800,``
``                    "image_size_y": 600,``
``                    "fov": 90.0``
``}``

Now, if I run the carla simulator and ros-bridge, and start the ego_vehicle at x=150, y=-130 (/map frame coordinates), and check the transformations with

``ros2 run tf2_ros tf2_echo map ego_vehicle``
``At time 9.540466212``
``-Translation: [150.000, -133.000, 0.001]``
``-Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]``

which is fine.

Now, the transformation for the 'rgb_front' is

``    ros2 run tf2_ros tf2_echo ego_vehicle ego_vehicle/rgb_front  ``
``    At time 10.440466226  ``
``    -Translation: [0.000, 0.000, 0.000]  ``
``    -Rotation: in Quaternion [-0.500, 0.500, -0.500, 0.500]  ``
``

which is not correct. What's more, if we start the simulator and the ros-bridge again, and start the vehicle at the same position as before (x=150, y=-133)

``ros2 run tf2_ros tf2_echo ego_vehicle ego_vehicle/rgb_front``
``At time 31.666939977``
``-Translation: [-14.213, -87.714, -0.285]``
``-Rotation: in Quaternion [-0.707, -0.000, 0.000, 0.707]``

which certainly isn't correct. Restarting the simulator and ros-bridge multiple times, the output of 'ros2 run tf2_ros tf2_echo ego_vehicle ego_vehicle/rgb_front' changes for each run. Shouldn't the transform ego_vehicle -> ego_vehicle/rgb_front be the offset between the vehicle body and the sensor, which we specify in the objects.json? Now it seems like the ego_vehicle -> ego_vehicle/sensor transformations take on random values. I have no idea on how to start fixing this, though.

Also, /tf_static is still not publishing the static transforms between 'ego_vehicle' and 'ego_vehicle/sensor' frames, as pointed out in #338. Based on discussion in #23, I understand publishing the relationships between sensors and vehicle frame though /tf_static would be the desired behavior .

KevinLADLee, 2021-09-17:
After reviewing the code, I think this problem is relative to the 'relative_spawn_pose' variable. So I changed the actor_factory, make it directly get from the sensor definition file.  It seems to work for me. Could you please help me to confirm that this is fixing the issue? @Roosstef  @lasuomela

KevinLADLee, 2021-09-17:
Hi, @lasuomela . I have also tested the code by running the following commands.
```
# For ros1
roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch 
# For ros2
ros2 launch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch.py
```
The first issue you mentioned about camera sensor rotation is not wrong. Because the definition of the camera coordinate system is different from lidar, GNSS, and IMU sensors in ROS. For more information about this, you could have a look at #511  and #516 .

Also, for `rgb_view` camera, the output of tf is:
```
- Translation: [-4.500, 0.000, 2.800]
- Rotation: in Quaternion [-0.579, 0.579, -0.406, 0.406]
```
I think this output is correct. Maybe there is some other config file in your workspace or launch file?
[image]

lasuomela, 2021-09-20:
@KevinLADLee, yes it seems that the different orientation of the camera coordinate system was the cause of the orientation difference. And you're probably right about the `rgb_view` issue.

So I can confirm the fix is working, nice job! Thanks.