carla-simulator/ros-bridge #566: Supports Carla 0.9.12?

rowandempster, 2021-08-23:
Does the bridge work with the latest https://github.com/carla-simulator/carla/releases/tag/0.9.12 release? I'm planning on manually updating the https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/src/carla_ros_bridge/CARLA_VERSION file and seeing what breaks, wondering if anyone else has already tried :)

kwea123, 2021-08-24:
I just tried (ros2). I did a couple of modifications to make it work:
1. Like you said, modify the version number in that file to `0.9.12`.
2. add `import cv2` in `camera.py` just before `from cv_bridge import CvBridge`.
3. increase the timeout with `ros2 launch carla_ros_bridge carla_ros_bridge.launch.py timeout:=30` as in #361.
The topics seem to be correctly publishing (seen with `ros2 topic echo `, I am still working with rviz visualization.

hofbi, 2021-08-25:
I already use the ros bridge with 0.9.12. So far, I just had to update the version file, but I did not try with all features. I created a PR for this and wait for getting it merged: #563