carla-simulator/ros-bridge #583: Create Official Docker Image

Bonifatius94, 2021-11-05:
Hi,

I've seen that you provide a Dockerfile for installing the carla_ros_bridge into the CARLA simulation environment which is really nice. The only downside is that there's no registry where I can download that image and building it myself did run into some compilation errors.

So, I'm really wondering whether the Dockerfile in the ./docker directory actually works or did ever work. Doesn't seem so, at least for the tags 0.9.10.1 and 0.9.11 I've checked out and tried to build.

Am I doing something wrong? How is this ./docker/build.sh script supposed to work? I've tried to build for the predefined ROS version and for the version "noetic" (my actual target).

The error message I'm always encountering is the following. It's thrown when calling "catkin_make install".

```text
--- stderr: carla_ackermann_msgs
CMake Error at CMakeLists.txt:41 (find_package):
  By not providing "Findcarla_msgs.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "carla_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "carla_msgs" with
  any of the following names:

    carla_msgsConfig.cmake
    carla_msgs-config.cmake

  Add the installation prefix of "carla_msgs" to CMAKE_PREFIX_PATH or set
  "carla_msgs_DIR" to a directory containing one of the above files.  If
  "carla_msgs" provides a separate development package or SDK, be sure it has
  been installed.

---
Failed   <<< carla_ackermann_msgs [1.22s, exited with code 1]
```

Thanks for your help in advance.

Bonifatius94, 2021-12-02:
I've created my own little Docker image that actually works for launching scenarios with RVIZ + scenario runner, etc. I'll publish it in a few months once my internal project is over.

MTDzi, 2022-01-23:
@Bonifatius94 did you have any spare time to publish this?
It doesn't need to be pretty ;)

Bonifatius94, 2022-01-24:
You know, I'm in a competition that's still ongoing until the end of March, but I promise I have some goodies for you ^^

It's an Image that's based on *osrf/ros:noetic-desktop-full-focal*. It allows you to specify vehicle sensors, set up a custom RVIZ layout, spawn NPCs and adjust various settings like car spawn settings, world settings, XOSC scenarios, etc.

So, please remind me in 3 months and I'll be happy to share my Dockerfile with the world :D

MTDzi, 2022-01-24:
Awesome, thanks for the reply, I've just marked a date in my calendar ;)

And since you've been this responsive, I'll ask another question: do you have knowledge / opinion about how well CARLA works with the ROS2 bridge?

I tried running it with the ROS1 bridge > year ago and there were some glitches (there was a weird jerkiness to the car) so I've just started gathering intel on whether I should go with CARLA + ROS2-bridge, or use the LGSVL (for which the ROS2 bridge works, as far as I know).

I'm just curious what

Bonifatius94, 2022-01-24:
Sorry to disappoint you, but I'm not at all into ROS 2, so can't give you any advice on that one.

And about this jerkiness. I have only tried CARLA 0.9.10.1 and that one works somewhat fine. You can set up the ROS bridge to work timestep-based synchronous mode in case your graphics are too bad. At my project, we're using a NVIDIA 3080 Ti, so this could really be the issue, depending on the amount of sensors you're trying to use for perception tasks.

Another interesting point is the Ackermann PID controller which is set up to react very delayed. We're dealing with it by sending 10 driving signals / s and also the PID settings had to be customized a bit to let the changes through more quickly.

Hope this helps, although it is a bit off-topic :)
Best wishes, Marco

MTDzi, 2022-01-24:
Thanks!
I just realized I cut my last sentence short, but I guess you've figured out how I was going to end it ;)

All the best in the competition you're participating!

Bonifatius94, 2022-01-24:
Thanks man. See you in 3 months :)