carla-simulator/ros-bridge #584: [Windows10] colcon build failed

Sylvia123456, 2021-11-09:
Hi, I wonder if ros-bridge can work on Windows. So I installed ROS 2 Foxy (ros2-foxy-20211013-windows-release-amd64 ), ros2 can work, and run the demo well. Then follow the instruction from this website (https://carla.readthedocs.io/projects/ros-bridge/en/latest/ros_installation_ros2/)  to install the ros-bridge. but I got some problems.

1、when  running `rosdepc install --from-paths src --ignore-src -r`, the console output some errors. I am not sure if it success or not? 
[image]

2、when running `colcon build` , it always failed at carla_ros_scenario_runner_types
[image]

Could anyone give me some help ? Many Thanks!!!!!

Sylvia123456, 2021-11-10:
I know the reason , the file path is too long ,more then 260 characters.....already fixed.  I will close this issue

closed 2021-11-10 by Sylvia123456

smp4966, 2022-07-05:
Hi, I installed ROS 2 Foxy and its work perfectly 

Then follow the instruction from this website (https://carla.readthedocs.io/projects/ros-bridge/en/latest/ros_installation_ros2/) to install the ros-bridge.

1)Install ROS:  For Ubuntu 20.04 (Focal)(Debian packages)(https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Debians.html)
2)CARLA 0.9.13 bulid with UE4.26.2  (python 3.8)

mkdir -p ~/carla-ros-bridge && cd ~/carla-ros-bridge
git clone --recurse-submodules https://github.com/carla-simulator/ros-bridge.git src/ros-bridge
Set up the ROS environment:source /opt/ros/foxy/setup.bash
rosdep update
rosdep install --from-paths src --ignore-src -r

but I got some problems.
when running colcon build , it always failed at carla_ros_scenario_runner_types

Starting >>> carla_msgs
Starting >>> ros_compatibility
Starting >>> carla_common
Starting >>> carla_ros_scenario_runner_types
Starting >>> carla_waypoint_types
Starting >>> carla_twist_to_control
Starting >>> rqt_carla_control
Finished <<< carla_common [2.00s]                               
Finished <<< ros_compatibility [2.05s]
Finished <<< carla_twist_to_control [2.02s]
Finished <<< rqt_carla_control [2.05s]                              
--- stderr: carla_waypoint_types                                    
/usr/bin/ld: /usr/local/lib/libpython3.8.a(obmalloc.o): relocation R_X86_64_PC32 against symbol `stderr@@GLIBC_2.2.5' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/carla_waypoint_types__python.dir/build.make:148: rosidl_generator_py/carla_waypoint_types/libcarla_waypoint_types__python.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:301: CMakeFiles/carla_waypoint_types__python.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< carla_waypoint_types [2.71s, exited with code 2]
Aborted  <<< carla_ros_scenario_runner_types [2.77s]
Aborted  <<< carla_msgs [2.80s]
                                 
Summary: 4 packages finished [3.07s]
  1 package failed: carla_waypoint_types
  2 packages aborted: carla_msgs carla_ros_scenario_runner_types
  3 packages had stderr output: carla_msgs carla_ros_scenario_runner_types carla_waypoint_types
  12 packages not processed

Thank you in advance

[image]