carla-simulator/ros-bridge #587: Add new Scenario

kargarisaac, 2021-11-12:
Hi,
I'm trying to use another scenario from scenario runner with carla_ad_demo. I put the scenario file `LaneChangeSimple.xosc` in the config folder next to the `FollowLeadingVehicle.xosc` file and change both launch files in the launch folded. Then I build again and source and run the `ros2 launch carla_ad_demo carla_ad_demo_with_scenario.launch.py` command. I can see the new added scenario in rviz but when I click on the execute button, it doesn't work. I see a warning in the terminal that:

```
Requested scenario file not existing /home/isaac/carla-ros-bridge/install/carla_ad_demo/share/carla_ad_demo/config/PedestrianCrossingFront.xosc
```

It seems the file that I put in the config folder, is not generated in the install folder (carla-ros-bridge/install). I tested removing install, build, and log folders and build again, but it didn't work.

Am I doing sth wrong? Or how should I do the available openscenario files from scenario runner to carla ros bridge?

Thank you

kargarisaac, 2021-12-17:
It seems that the problem was with changing the city. When I add a scenario from town01, it works. Any suggestions on how to fix this? where should I look for the problem?

EnnaSachdeva, 2022-11-03:
Hi, 
Is there any update on this issue yet? Did you solve this issue?