carla-simulator/ros-bridge #591: Windows - timeout connecting to Carla

filipthoen2, 2021-12-10:
Hello ROS2 Community!

I want to use Carla ros-bridge on Windows (I understand this is experimental, but I can't move to Linux) -- I got ros-bridge & all its dependencies all compiled (that was an adventure!), but when I run ros-bridge it fails to connect to Carla & times out.

I'm using a Windows Command Prompt with Administrator permissions & launch Carla & ros-bridge as follows (I'm using the same terminal for both):

C:\carla\bridge\src\carla_ros_bridge\launch>c:\carla\carla\WindowsNoEditor\CarlaUE4.exe

C:\carla\bridge\src\carla_ros_bridge\launch>ros2 launch -d carla_ros_bridge carla_ros_bridge.launch.py

which shows the following output:

[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[INFO] [launch]: All log files can be found below C:\Users\filip\.ros\log\2021-12-10-11-15-08-138231-filip-5540-28608
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x000001C6D2872E80>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x000001C6D2872E80>' ? '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x000001C6D2872A00>'
Executing <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at c:\carla\ros2-foxy-release\Lib\site-packages\launch\launch_service.py:226> result=None created at c:\carla\ros2-foxy-release\Lib\site-packages\launch\launch_service.py:314> took 0.390 seconds
[INFO] [bridge.EXE-1]: process started with pid [7968]
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessStarted'
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x000001C6D27C93A0>'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x000001C6D2C06DF0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x000001C6D2C06DF0>' ? '<launch.event_handlers.on_process_io.OnProcessIO object at 0x000001C6D2C06640>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x000001C6D2C49430>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x000001C6D2C49430>' ? '<launch.event_handlers.on_process_io.OnProcessIO object at 0x000001C6D2C06640>'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x000001C6D23099D0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x000001C6D23099D0>' ? '<launch.event_handlers.on_process_io.OnProcessIO object at 0x000001C6D2C06640>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x000001C6D2C49310>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x000001C6D2C49310>' ? '<launch.event_handlers.on_process_io.OnProcessIO object at 0x000001C6D2C06640>'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x000001C6D2C06DF0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x000001C6D2C06DF0>' ? '<launch.event_handlers.on_process_io.OnProcessIO object at 0x000001C6D2C06640>'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x000001C6D2C069A0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x000001C6D2C069A0>' ? '<launch.event_handlers.on_process_io.OnProcessIO object at 0x000001C6D2C06640>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x000001C6D2C06DF0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x000001C6D2C06DF0>' ? '<launch.event_handlers.on_process_io.OnProcessIO object at 0x000001C6D2C06640>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x000001C6D23099D0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x000001C6D23099D0>' ? '<launch.event_handlers.on_process_io.OnProcessIO object at 0x000001C6D2C06640>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x000001C6D2C49340>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x000001C6D2C49340>' ? '<launch.event_handlers.on_process_io.OnProcessIO object at 0x000001C6D2C06640>'
[ERROR] [bridge.EXE-1]: process has died [pid 7968, exit code 1, cmd 'c:\carla\bridge\install\carla_ros_bridge\lib\carla_ros_bridge\bridge.EXE --ros-args -r __node:=carla_ros_bridge --params-file C:\Users\filip\AppData\Local\Temp\launch_params_9znw7dhr --params-file C:\Users\filip\AppData\Local\Temp\launch_params_p44a299q --params-file C:\Users\filip\AppData\Local\Temp\launch_params_qyiov1c9 --params-file C:\Users\filip\AppData\Local\Temp\launch_params_pe7kyzgz --params-file C:\Users\filip\AppData\Local\Temp\launch_params_1uf_nekv --params-file C:\Users\filip\AppData\Local\Temp\launch_params_d3y1lg4z --params-file C:\Users\filip\AppData\Local\Temp\launch_params_de33obrm --params-file C:\Users\filip\AppData\Local\Temp\launch_params_wbxkjtgv --params-file C:\Users\filip\AppData\Local\Temp\launch_params_f12h0s9q --params-file C:\Users\filip\AppData\Local\Temp\launch_params_e1e_1xz6 --params-file C:\Users\filip\AppData\Local\Temp\launch_params_euwrpt01'].
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessExited'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x000001C6D27C9430>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x000001C6D27C9430>' ? '<launch.event_handlers.on_process_exit.OnProcessExit object at 0x000001C6D2C06D30>'
[bridge.EXE-1] [INFO] [1639163712.220253000] [carla_ros_bridge]: Trying to connect to localhost:2000

[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x000001C6D27C9430>' ? '<launch.event_handlers.on_process_exit.OnProcessExit object at 0x000001C6D2C06AC0>'
[INFO] [launch]: process[bridge.EXE-1] was required: shutting down launched system
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.Shutdown'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x000001C6D2BF6790>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x000001C6D2BF6790>' ? '<launch.event_handlers.on_shutdown.OnShutdown object at 0x000001C6D2C06C10>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x000001C6D2BF6790>' ? '<launch.event_handlers.on_shutdown.OnShutdown object at 0x000001C6D2872BE0>'

I checked the log file but it doesn't produce any additional info. I tried to manually launch the command being launched above by ROS:

C:\carla\bridge\src\carla_ros_bridge\launch>c:\carla\bridge\install\carla_ros_bridge\lib\carla_ros_bridge\bridge.EXE --ros-args -r __node:=carla_ros_bridge --params-file C:\Users\filip\AppData\Local\Temp\launch_params_9znw7dhr --params-file C:\Users\filip\AppData\Local\Temp\launch_params_p44a299q --params-file C:\Users\filip\AppData\Local\Temp\launch [... cut ...]
- checked the TCP/IP ports in use using 'netstat -ab'. Ports 2000 & 2001 are assigned to 'CarlaUE4-Win64-Shipping.exe' & in "Listening" mode
- tried to run Carla with '--carla-server' option.

Info on setup below. Any detail on how to further troubleshoot this? 

Carla itself doesn't seem to have a log file, which could give some clue from the server side why the connection is not going through.

Thanks in advance,

Filip

Setup:
- Carla: v0.9.13
- Python: v3.8.0
- ROS2: Foxy (ros2-foxy-20211013-windows-release-amd64.zip)
- Opencv-3.4.6-vc16.VS2019.zip
- Host: Windows 10
- Compiler: MSVC 2019

Jags27GitHub, 2022-06-22:
I also arrived at the same error. In the doc, it is mentioned clearly that ros bridge is not tested on windows. Would like to know if any update on this feature or any workarounds are appreciated.

ronyshaji, 2023-06-26:
@filipthoen2 Did you found out a solution to this problem ?

jongminj, 2023-10-30:
@filipthoen2  @ronyshaji 
hello, Did you find out a solution to this problem?

filipthoen2, 2023-10-30:
Unfortunately, no, I haven't been able to find a solution.

jongminj, 2023-10-30:
@filipthoen2 
Now I have just found solution.
I was using ROS_noetic on WSL2 and  carla_0.9.13 for window version.
I changed the host IP from 'localhost' to my local IP in the launch file and set the timeout parameter to 200.
It worked for me