carla-simulator/ros-bridge #593: support for Carla 0.9.13

kargarisaac, 2021-12-17:
Hi,

Any plan to support Carla v0.9.13?

Thanks

AndyYangjd, 2022-01-12:
You can change the CARLA_VERSION file in the pkg of carla_ros_bridge to 0.9.13，so all is right.
I can simulate  in localhost machine with the above methods, but can't launch it with tcp.

Chuck1989niu, 2022-03-03:
> You can change the CARLA_VERSION file in the pkg of carla_ros_bridge to 0.9.13，so all is right. I can simulate in localhost machine with the above methods, but can't launch it with tcp.

您好，问一下。在本地主机上怎么打开ROS2。我也用的是Carla0.9.13，无法通过  ros2 launch carla_ros_bridge carla_ros_bridge.launch.py 这个指令打开ROS2。  期待您的答复。非常感谢！

AndyYangjd, 2022-03-07:
> 您好，问一下。在本地主机上怎么打开ROS2。我也用的是Carla0.9.13，无法通过 ros2 launch carla_ros_bridge carla_ros_bridge.launch.py 这个指令打开ROS2。 期待您的答复。非常感谢！

 I can't understand your question about how to launch ros2. If this, please cite on [Ros2-Foxy](https://docs.ros.org/en/foxy/index.html). 我不是很清楚你所要表达的问题是否是关于如果启动Ros2，如果是这样的话，请参考Ros2官方文档[Ros2-Foxy](https://docs.ros.org/en/foxy/index.html)。

Chuck1989niu, 2022-03-08:
> I can't understand your question about how to launch ros2. If this, please cite on [Ros2-Foxy](https://docs.ros.org/en/foxy/index.html). [...]

I wanted to test Carla0.9.13 with ROS2 and ROS bridge. The CARLA part is running correctly.Start the ROS bridge properly.(ros2 launch carla_ros_bridge carla_ros_bridge.launch.py), however when I try to launch Spawn objects, I get an error that one of the parameters is None.

My environment:

Ros 2 Foxy
Carla0.9.13 on brach master 
Latest ROS bridge

Execution command :
The CARLA part:
~$ cd carla   
~$ make launch  (play)

The ROS2 part:
~$ cd carla-ros-bridge
~$ source ./install/setup.bash
~$ ros2 launch carla_ros_bridge carla_ros_bridge.launch.py timeout:=30
In another terminal
~$ ros2 launch carla_spawn_objects carla_spawn_objects.launch.py

The full exception:

[INFO] [launch]: All log files can be found below /home/chuck/.ros/log/2022-03-08-09-55-46-690946-chuck01-10518
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=TypeError("Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class 'NoneType'>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)")>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 1 more time]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 444, in execute
    self._perform_substitutions(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 399, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 160, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 95, in evaluate_parameter_dict
    raise TypeError(
TypeError: Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class 'NoneType'>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)

How can this error be resolved?

martino92, 2022-04-12:
> You can change the CARLA_VERSION file in the pkg of carla_ros_bridge to 0.9.13，so all is right. I can simulate in localhost machine with the above methods, but can't launch it with tcp.

Where is the CARLA VERSION file? Would you mind give a specific path to the file?

wenxuanou, 2022-04-25:
> > You can change the CARLA_VERSION file in the pkg of carla_ros_bridge to 0.9.13，so all is right. I can simulate in localhost machine with the above methods, but can't launch it with tcp.
> 
> Where is the CARLA VERSION file? Would you mind give a specific path to the file?

here is the file ~/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/CARLA_VERSION