carla-simulator/ros-bridge #594: Vehicles spawned by generate_traffic.py file do crazy behaviors

kargarisaac, 2021-12-17:
Hi,

I'm trying to spawn some cars in the city using the `generate_traffic.py` file but cars do crazy movements, go out of the road, and they crash with each other. I use Carla 0.9.12 and the respective ros-bridge. Does anyone have a guess where the problem may be?

Thanks

Cypreis, 2023-06-23:
Hello,

I'm having the same issues, using ROS2 Foxy and Ubuntu 20.04, vehicles just run into walls and go out of the road constantly, tried to re-activate autopilot after spawning, doesn't work, synchronous mode is on.

etfini, 2023-09-20:
I just found that don't set the traffic manager to sychronous mode it will work. So just comment out the line `traffic_manager.set_synchronous_mode(True)` and then it worked for me.