carla-simulator/ros-bridge #601: ROS2 Ego Vehicle Example crashes.

amansrf, 2022-02-14:
When trying to install the ros-bridge for ros2 foxy following this [guide](https://carla.readthedocs.io/projects/ros-bridge/en/latest/ros_installation_ros2/), the ego vehicle ros-bridge example crashes 
`ros2 launch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch.py `

I get the following error on terminal:

[image]

System Specs:
OS: Ubuntu 20.04
ROS: ROS2-Foxy
Carla & Rosbridge Version: 0.9.12

Jia-Wei-Wang, 2022-02-14:
System Specs:
OS: Ubuntu 20.04
ROS: Release -foxy-20220208
Carla Simulator&ROS bridge: tags/0.9.11

After installing  [<Release -foxy-20220208>](https://github.com/ros2/ros2/tree/release-foxy-20220208), I encountered the same errors when I want to launch ```carla_spawn_objects```. Main program `carla_ros_bridge` and `carla_manual_control` can be launched successfully with parsing input arguments, e.g. `town:="Town03" fixed_delta_seconds:=0.05`.

My temporary solution is to run python script `carla_spawn_objects.py ` and set the input argument to hard-coded`objects_definition_file` = '/path/to/cfg.json' and It works.

amansrf, 2022-02-14:
I tried your suggestion by running the following command after starting the `carla_ros_bridge` , however I still received the same error.

Exact steps I took:
1. Started the Carla Server ( `./CarlaUE4.sh`)
2. sourced carla-ros-bridge workspace and then ran `ros2 launch carla_ros_bridge carla_ros_bridge.launch.py` succesfully
3. In a new terminal, I tried to launch the carla_spawn_objects launchfile using the following:
`ros2 launch install/carla_spawn_objects/share/carla_spawn_objects/carla_spawn_objects.launch.py objects_definition_file:=install/carla_spawn_objects/share/carla_spawn_objects/config/objects.json`

I get a similar error as before. Did I choose the wrong launchfile or wrong object file?

wuxiaohua1011, 2022-02-17:
Short answer: 
Simply provide name(i think the carla team intended the `spawn_point_ego_vehicle_value` to be filled with name?) for ego vehicle 
`ros2 launch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch.py spawn_point_ego_vehicle:="spawn_point_hero0"` 

and then you can use RVIZ to see the camera data streams and stuff. 

Long answer: 
So tracing back, I see that the `carla_ros_bridge_with_example_ego_vehicle.launch.py` calls `carla_spawn_objects.launch.py`, which the latter requires this field called `spawn_point_ego_vehicle`. 

Recommendation to CARLA team: maybe provide an actual default value instea of empty string for `spawn_point_ego_vehicle` field?