carla-simulator/ros-bridge #602: ROS Bridge error:  1 package had stderr output: pcl_recorder

linpanusst, 2022-02-24:
When I use ros bridge and carla. I faced some problems. When I just start the example on the official document. Run the command `ros2 launch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch.py` . I faced the error messages below. 

> Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=TypeError("Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class 'NoneType'>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)")>
...

> TypeError: Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class 'NoneType'>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)
...

> [ERROR] [bridge-1]: process[bridge-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [bridge-1]: sending signal 'SIGTERM' to process[bridge-1]
[ERROR] [bridge-1]: process has died [pid 21994, exit code -15, cmd '/home/franklinpan/carla-ros-bridge/install/carla_ros_bridge/lib/carla_ros_bridge/bridge --ros-args -r __node:=carla_ros_bridge --params-file /tmp/launch_params_39w9fnke --params-file /tmp/launch_params_2ch3pyft --params-file /tmp/launch_params_qpgy9uoj --params-file /tmp/launch_params_d9apit2s --params-file /tmp/launch_params_3z3lehz_ --params-file /tmp/launch_params_3_1iikfy --params-file /tmp/launch_params_sxatit15 --params-file /tmp/launch_params_czqh_66p --params-file /tmp/launch_params_u9yqsinr --params-file /tmp/launch_params_vb8bgwrf --params-file /tmp/launch_params_3ftkmck1'].
[INFO] [launch]: process[bridge-1] was required: shutting down launched system

My comupter information: OS: ubuntu 20.04, GPU: T2000, Cuda version: 11.4. GPU driver version: 470.103.01.
My software version: Carla: 0.9.12, Python: 3.8, Ros: foxy, ROS_bridge: 0.9.12 (latest).

Other information: when colcon build the ros_bridge package. The pcl_recorder will always failed. See the information below.

> --- stderr: pcl_recorder                                                                                     
** WARNING ** io features related to openni will be disabled
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled

> Summary: 19 packages finished [31.6s]
  1 package had stderr output: pcl_recorder

kapenon, 2022-02-25:
I hope this discussion would help you.
https://github.com/carla-simulator/ros-bridge/issues/601

ErikKrauter, 2022-11-28:
I am using Ubuntu 20.04 with ROS2 galactic. When colcon building the ros bridge workspace I encounter the same warning: 
```
Starting >>> carla_ackermann_control
Finished <<< carla_ackermann_control [1.04s]                                 
--- stderr: pcl_recorder                                                      
** WARNING ** io features related to openni will be disabled
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
---
Finished <<< pcl_recorder [19.1s]
Finished <<< rviz_carla_plugin [27.8s]                         
Starting >>> carla_ad_demo
Finished <<< carla_ad_demo [0.89s]                     

Summary: 19 packages finished [44.7s]
  1 package had stderr output: pcl_recorder
```

sangeetsu, 2023-10-16:
Similar issue

> --- stderr: pcl_recorder                                                                                     
** WARNING ** io features related to openni will be disabled
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
---
Finished <<< pcl_recorder [21.2s]
Finished <<< rviz_carla_plugin [30.6s]                         
Starting >>> carla_ad_demo
Finished <<< carla_ad_demo [2.20s]                     

Summary: 19 packages finished [58.6s]
  5 packages had stderr output: carla_ackermann_msgs carla_msgs carla_ros_scenario_runner_types carla_waypoint_types pcl_recorder

Anhar2, 2024-03-06:
> Similar issue
>
>
> Finished <<< pcl_recorder [21.2s] Finished <<< rviz_carla_plugin [30.6s] Starting >>> carla_ad_demo Finished <<< carla_ad_demo [2.20s]
>
> Summary: 19 packages finished [58.6s] 5 [...]

could you solve it?

komorekm, 2024-04-24:
I'm a bit late, but for me doing
```shell
sudo apt install ros-melodic-pcl-ros
sudo apt install pcl-tools
``` 
solved the problem.