carla-simulator/ros-bridge #603: Supports Carla 0.9.13?

Chuck1989niu, 2022-03-04:


Chuck1989niu, 2022-03-08:
I wanted to test Carla0.9.13 with ROS2 and ROS bridge. The CARLA part is running correctly.Start the ROS bridge properly.(ros2 launch carla_ros_bridge carla_ros_bridge.launch.py), however when I try to launch Spawn objects, I get an error that one of the parameters is None.

My environment:

Ros 2 Foxy
Carla0.9.13 on brach master 
Latest ROS bridge

Execution command :
The CARLA part:
~$ cd carla   
~$ make launch  (play)

The ROS2 part:
~$ cd carla-ros-bridge
~$ source ./install/setup.bash
~$ ros2 launch carla_ros_bridge carla_ros_bridge.launch.py timeout:=30
In another terminal
~$ ros2 launch carla_spawn_objects carla_spawn_objects.launch.py

The full exception:

[INFO] [launch]: All log files can be found below /home/chuck/.ros/log/2022-03-08-09-55-46-690946-chuck01-10518
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=TypeError("Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class 'NoneType'>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)")>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 1 more time]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 444, in execute
    self._perform_substitutions(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 399, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 160, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 95, in evaluate_parameter_dict
    raise TypeError(
TypeError: Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class 'NoneType'>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)

How can this error be resolved?

alexkutsan, 2022-04-30:
Same question from me :-) 
Not able to use 9.1.12 because of issue https://github.com/carla-simulator/carla/issues/3599 
But 9.1.13 not compatible with ROS bridge

FelipeGdM, 2022-06-27:
I had the same problem when trying to run ros-brigde with Carla 0.9.13. Running it with a modified version of `launch_ros` (https://github.com/ros2/launch_ros/pull/320) it was possible to discover that the problem is the evaluation of the parameter `spawn_point_ego_vehicle`.

I believe that if you pass any string for the parameter `spawn_point_ego_vehicle` in the command line, the error shouldn't occur. 

```
ros2 launch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch.py spawn_point_ego_vehicle:=spawn_point
```

If you pass an invalid string, the message should be

```
[carla_spawn_objects-4] [WARN] [1656355794.696531425] [carla_spawn_objects]: Invalid spawnpoint 'spawn_point'
[carla_spawn_objects-4] [WARN] [1656355794.697031216] [carla_spawn_objects]: ego_vehicle: Could not use 
spawn point from parameters, the spawn point from config file will be used.
```

And then ros bridge runs normally

berndgassmann, 2022-07-08:
Hi,
I found a different solution for the problem with the spawn_point by explicitly converting it to string after launch.substitution. See one of the launch files [here](https://github.com/carla-simulator/ros-bridge/pull/624/files#diff-8882441b250de2168a3be4b32a0cb6bcdcc03b918155a4cd1382bc1a269e59e5R75). Even if I am working on 0.9.11 branch, I  believe this solution should also work independent from the CARLA version (as it's related rather to the ROS2 version used).

andrejpadilha2, 2022-07-12:
> 

Worked for me @FelipeGdM, thank you/muito obrigado!