carla-simulator/ros-bridge #607: rosdep install --from-paths src --ignore-src -r  -- Failed to Install

SandyWare, 2022-03-17:
**Environment:** Ubuntu  18.0.4  (Bionic), ROS Melodic ,  CARLA  Simulator version 0.9.9.
**Issue:**   pip: command [sudo -H /usr/bin/python -m pip install -U transforms3d] failed

Steps  followed as per the below documents : 
https://carla.readthedocs.io/en/0.9.9/ros_installation/#bridge-installation

Command  : **rosdep install --from-paths src --ignore-src -r**

Setting up ros-melodic-carla-msgs (1.3.0-1bionic.20210505.011451) ...
/usr/bin/python: No module named pip
executing command [sudo -H /usr/bin/python -m pip install -U transforms3d]
/usr/bin/python: No module named pip
/usr/bin/python: No module named pip
ERROR: the following rosdeps failed to install
  **pip: command [sudo -H /usr/bin/python -m pip install -U transforms3d] failed**
  pip: Failed to detect successful installation of [transforms3d]

Any suggestion or help is appricated.

SandyWare, 2022-03-24:
I have installed CRALA 9.13 and built ROS Bridge with the latest version. This solves this issue.

closed 2022-03-24 by SandyWare

YoonjungChoi, 2022-04-02:
Hello. I am trying to follow this [deliverychallenge-repo](https://github.com/aws-samples/aws-robomaker-sample-application-delivery-challenge). Here are the detail steps but robomaker API is deprecated and concepts changed(Docker).

So I am trying to building container, but the in [Dockerfile](https://docs.aws.amazon.com/robomaker/latest/dg/development-build-sim-containers.html) , I got the same issue.

Do you have any idea about this?

martino92, 2022-04-12:
> 

Do you mean that the latest version of ROS bridge is master branch?