carla-simulator/ros-bridge #608: Carla 0.9.13 and ros-bridge  build failuer due to version compability

SandyWare, 2022-03-24:
**Environment :** 
Ubuntu 18.0.4
Carla version (tag: 0.9.13) latest.
Ros_bridge : latest

I have build ros_bridge successfully using catkin_make.
Then ros_bridge launched using 
    roslaunch carla_ros_bridge carla_ros_bridge.launch

Then I got the below error :
**CARLA python module version 0.9.12 required. Found: 0.9.13**
```
started core service [/rosout]
process[carla_ros_bridge-2]: started with pid [15130]
[INFO] [1648114186.893339, 0.000000]: Trying to connect to localhost:2000
[FATAL] [1648114186.897207, 0.000000]: CARLA python module version 0.9.12 required. Found: 0.9.13
[INFO] [1648114186.898706, 0.000000]: Shutting down...
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/core.py", line 572, in signal_shutdown
    h()
  File "/home/ubuntu/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py", line 349, in destroy
    self.shutdown.set()

```

How I can change version ros_bridge to make it compatible with CARLA 0.9.13 ? 
Any help suggestion is appricated.

miguelriemoliveira, 2022-03-28:
Hi @SandyWare ,

I have the same problem. Tried to rollback to 0.9.12 with command

sudo pip3 install carla==0.9.12 

but then when I run the bridge I get the error:

```
[WARN] [1648462252.676591, 0.000000]: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API. Client API version: 0.9.12. Simulator API version: 0.9.13
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.12 
WARNING: Simulator API version  = 0.9.13 
================================================================================REQUIRED process [carla_ros_bridge-2] has died!
process has died [pid 25138, exit code -11, cmd /home/mike/catkin_ws/src/carla-ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py __name:=carla_ros_bridge __log:=/home/mike/.ros/log/58bdc990-ae7f-11ec-8c95-970962904cc0/carla_ros_bridge-2.log].
log file: /home/mike/.ros/log/58bdc990-ae7f-11ec-8c95-970962904cc0/carla_ros_bridge-2*.log
Initiating shutdown!
================================================================================
```

leon-D-chaser, 2022-04-21:
@wzr6009 I tried your method, but i still got an Error.

`[ERROR] [1650535527.868848, 0.000000]: Error: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000`

bilibilicharger, 2022-04-23:
> @wzr6009 I tried your method, but i still got an Error.
> 
> `[INFO] [1650536178.043873, 0.000000]: Loading town 'Town01' (previous: 'Carla/Maps/Town10HD_Opt'). [ERROR] [1650535527.868848, 0.000000]: Error: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000 [INFO] [1650535527.872969, 0.000000]: Shutting down... `
did you solve this problem？

wzr6009, 2022-04-30:
can your ping your carla server ip, i used carla lasted , ros-bridge can run

alexkutsan, 2022-04-30:
Commenting this exit gives an ability to run ros_bridge against carla 0.9.13
But there are still issues in some use cases 
 : 
 ```
 Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 5 more times]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 444, in execute
    self._perform_substitutions(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 399, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 160, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 95, in evaluate_parameter_dict
    raise TypeError(
TypeError: Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class 'NoneType'>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)
[INFO] [bridge-1]: process started with pid [748]
Task exception was never retrieved
future: <Task finished name='Task-8' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=AttributeError("'NoneType' object has no attribute 'complete'")>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/opaque_function.py", line 75, in execute
    return self.__function(context, *self.__args, **self.__kwargs)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 444, in __on_signal_process_event
    if self._subprocess_protocol.complete.done():
AttributeError: 'NoneType' object has no attribute 'complete'
[bridge-1] [INFO] [1651329416.432205928] [carla_ros_bridge]: Trying to connect to 192.168.0.191:2000
[bridge-1] [FATAL] [1651329416.434158972] [carla_ros_bridge]: CARLA python module version 0.9.12 required. Found: 0.9.13
[bridge-1] [INFO] [1651329416.500372232] [carla_ros_bridge]: Loading town 'Town01' (previous: 'Carla/Maps/Town01').
[bridge-1] [INFO] [1651329418.453280442] [carla_ros_bridge]: synchronous_mode: True
[bridge-1] [INFO] [1651329418.453722887] [carla_ros_bridge]: fixed_delta_seconds: 0.05
[bridge-1] [INFO] [1651329418.517162325] [carla_ros_bridge]: Parameters:
[bridge-1] [INFO] [1651329418.517514806] [carla_ros_bridge]:   host: 192.168.0.191
[bridge-1] [INFO] [1651329418.518062734] [carla_ros_bridge]:   port: 2000
[bridge-1] [INFO] [1651329418.518533631] [carla_ros_bridge]:   timeout: 2
[bridge-1] [INFO] [1651329418.519057009] [carla_ros_bridge]:   passive: False
[bridge-1] [INFO] [1651329418.519520514] [carla_ros_bridge]:   synchronous_mode: True
[bridge-1] [INFO] [1651329418.520075558] [carla_ros_bridge]:   synchronous_mode_wait_for_vehicle_control_command: False
[bridge-1] [INFO] [1651329418.520591692] [carla_ros_bridge]:   fixed_delta_seconds: 0.05
[bridge-1] [INFO] [1651329418.521066700] [carla_ros_bridge]:   register_all_sensors: True
[bridge-1] [INFO] [1651329418.521568386] [carla_ros_bridge]:   town: Town01
[bridge-1] [INFO] [1651329418.595357114] [carla_ros_bridge]: Created TrafficLight(id=546)
[... cut ...] ams-file /tmp/launch_params_dtd1g55p --params-file /tmp/launch_params_5g_y67d7 --params-file /tmp/launch_params_1khvqdi5 --params-file /tmp/launch_params__uvuz44x --params-file /tmp/launch_params_c8urw1pd --params-file /tmp/launch_params_j3ivsvjm --params-file /tmp/launch_params_x2av50h_ --params-file /tmp/launch_params_uqpoam46 --params-file /tmp/launch_params_oii0fbkt --params-file /tmp/launch_params_42i7bls7'].
[INFO] [launch]: process[bridge-1] was required: shutting down launched system
```

kk-12138, 2022-06-01:
just change `ros2_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/CARLA_VERSION` to 0.9.13, colcon build and run again

smp4966, 2022-07-29:
what you can do is , Go to path  ros2_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/carla_version change line 0.9.12 to 0.9.13, run again colcon build. I hope it will work for you. please close this issue if you able to build successfully.

sandipware, 2022-08-02:
I have moved to docker version of CARLA  0.9.9, which fix my ros-bridge issue.

huetmathieu, 2023-03-17:
> @wzr6009 I tried your method, but i still got an Error.
> `[INFO] [1650536178.043873, 0.000000]: Loading town 'Town01' (previous: 'Carla/Maps/Town10HD_Opt'). [ERROR] [1650535527.868848, 0.000000]: [...]

Did you solve this problem too?

Kvvvv-yzk, 2024-09-08:
我也是这样