carla-simulator/ros-bridge #611: ROS2 failed to build carla_ad_demo

gunnxx, 2022-04-06:
The error is
```
--- stderr: carla_ad_demo                          
error: Multiple top-level packages discovered in a flat-layout: ['launch', 'config', 'resource'].

To avoid accidental inclusion of unwanted files or directories,
setuptools will not proceed with this build.

If you are trying to create a single distribution with multiple packages
on purpose, you should not rely on automatic discovery.
Instead, consider the following options:

1. set up custom discovery (`find` directive with `include` or `exclude`)
2. use a `src-layout`
3. explicitly set `py_modules` or `packages` with a list of names

To find more information, look for "package discovery" on setuptools docs.
```

I am using carla-simulator=0.9.12 and ros2 foxy. Anyone having the same issue? I am not familiar with ros2 commands such as colcon build.

joe28965, 2022-04-14:
Yes, got it just now. Have you been able to resolve it?

gunnxx, 2022-04-14:
I switched to `ros1 noetic` lmao.

joe28965, 2022-04-14:
Thanks, not going to work for me I'm afraid. Will try some stuff myself

joe28965, 2022-04-14:
Fixed it btw, will create a PR for it, if you're still interested

closed 2022-06-01 by joel-mb

muddassir93, 2025-12-03:
> The error is
> 
> ```
> --- stderr: carla_ad_demo                          
> error: Multiple top-level packages discovered in a flat-layout: ['launch', 'config', 'resource'].
> 
> To avoid accidental inclusion of unwanted files or directories,
> setuptools will not proceed with this build.
> 
> If you are trying to create a single distribution with multiple packages
> on purpose, you should not rely on automatic discovery.
> Instead, consider the following options:
> 
> 1. set up custom discovery (`find` directive with `include` or `exclude`)
> 2. use a `src-layout`
> 3. explicitly set `py_modules` or `packages` with a list of names
> 
> To find more information, look for "package discovery" on setuptools docs.
> ```
> 
> I am using `carla-simulator=0.9.12`, `ros2 foxy`, and `setuptools=62.0.0`. Anyone having the same issue? I am not familiar with ros2 commands such as colcon build.

This query is old and has been resolved. I am commenting for future reference.

This issue `error: Multiple top-level packages discovered in a flat-layout: ['share', 'resource'].` can be resolved if `packages=[]` is passed in `setup(...)`.

I got into this issue and resolved it with the above solution in `setuptools==68.1.2` and ros `humble`

PR #613 (2022-06-01): updated ad_demo to cmake over python to fix #611
As the title says, ad_demo fails in ROS 2 Foxy due to some python issue.

The thing is, there is no python used in that launch file. It's some configs and launch files. I'm no expert at python (wouldn't even call myself a novice tbh), so I changed the CMakeLists and package.xml to use ament_cmake instead.

It's a simple fix. If there is a desire to keep using ament_python (not sure why it would be mandatory) LMK. However I won't be able to fix #611 in python.

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/ros-bridge/ [... cut]