carla-simulator/ros-bridge #619: bridge does not work with rclpy 1.9.1

zernchri, 2022-06-14:
The ros-bridge stopped working for me, since I did apt upgrade on my PC. I get the following output messages when running `ros2 launch carla_ros_bridge carla_ros_bridge.launch.py`:

`[INFO] [launch]: All log files can be found below /home/sim/.ros/log/2022-06-13-19-15-10-143712-sim-SYS-7049GP-TRT-26978
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [bridge-1]: process started with pid [26980]
[bridge-1] Traceback (most recent call last):
[bridge-1]   File "/home/sim/carla-ros-bridge/install/carla_ros_bridge/lib/carla_ros_bridge/bridge", line 11, in <module>
[bridge-1]     load_entry_point('carla-ros-bridge==0.0.0', 'console_scripts', 'bridge')()
[bridge-1]   File "/home/sim/carla-ros-bridge/install/carla_ros_bridge/lib/python3.8/site-packages/carla_ros_bridge/bridge.py", line 385, in main
[bridge-1]     carla_bridge = CarlaRosBridge()
[bridge-1]   File "/home/sim/carla-ros-bridge/install/carla_ros_bridge/lib/python3.8/site-packages/carla_ros_bridge/bridge.py", line 63, in __init__
[bridge-1]     super(CarlaRosBridge, self).__init__("ros_bridge_node")
[bridge-1]   File "/home/sim/carla-ros-bridge/install/ros_compatibility/lib/python3.8/site-packages/ros_compatibility/node.py", line 138, in __init__
[bridge-1]     super(CompatibleNode, self).__init__(
[bridge-1]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/node.py", line 216, in __init__
[bridge-1]     self.declare_parameters(
[bridge-1]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/node.py", line 487, in declare_parameters
[bridge-1]     raise ParameterAlreadyDeclaredException(parameters_already_declared)
[bridge-1] rclpy.exceptions.ParameterAlreadyDeclaredException: ('Parameter(s) already declared', ['use_sim_time'])
[ERROR] [bridge-1]: process has died [pid 26980, exit code 1, cmd '/home/sim/carla-ros-bridge/install/carla_ros_bridge/lib/carla_ros_bridge/bridge --ros-args -r __node:=carla_ros_bridge --params-file /tmp/launch_params__k9_46r0 --params-file /tmp/launch_params_yoxtrkjv --params-file /tmp/launch_params_rgmanzbg --params-file /tmp/launch_params_zbyq4qj1 --params-file /tmp/launch_params_lt6787t2 --params-file /tmp/launch_params_o5hst251 --params-file /tmp/launch_params_09gel_2k --params-file /tmp/launch_params_q8t_dj29 --params-file /tmp/launch_params_33qgwaqm --params-file /tmp/launch_params_b6o8nu81'].
[INFO] [launch]: process[bridge-1] was required: shutting down launched system`

I think the problem occurs due to a version upgrade of rclpy from 1.9.0 to 1.9.1 and a lack of compatibility to the bridge. I think the reason for this is in `/home/sim/carla-ros-bridge/install/ros_compatibility/lib/python3.8/site-packages/ros_compatibility/node.py` in the `init` function (ln 136ff), where the use_sim_time variable is used. When changing the boolean values there, the error changes to `Error: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000`. Since carla runs properly, I think this error also occurs due to a missmatch of the sim_time.

Has someone else face the same issue and knows how to fix this? If I understand this errors correct, I would suggest everybody should have the same problem when changing the rclpy version.

zernchri, 2022-07-07:
The problem occurs due to a change in rclpy, as i expected. There was a change in the script node.py in rclpy, see https://github.com/ros2/rclpy/commit/8daacddc4fd72373fe9429de5ba3b2d2a5868078. The if automatically_declare_parameters_from_overrides part was changed. I solved that by simply changing this part of the node.py that changed in rclpy back to the old version. Now it works again for me.
To just set the automatically_declare_parameters_from_overrides to false in the ros_compatibilty node also worked to start the bridge (you also have to increase the timeout when starting the bridge), but then spawning of the ego_vehicle caused issues again. 
Dont know it changing the rclpy script is the cleanest solution, but at least it works at the moment.

fawadshahid, 2022-07-08:
@zernchri Thanks! I did the exact same thing before you posted this and it worked for me. I hope this release starts shipping with galactic.