carla-simulator/ros-bridge #620: ROS Bridge 0.9.10.1 failed to run test case with carla-0.9.10-py3.6-linux-x86_64.egg

SandyWare, 2022-06-22:
**Environment :**
Ubuntu 18.0.4
Carla version (tag: 0.9.10.1).
Ros_bridge : tag: 0.9.11  / 0.9.10.1  (I tried with both tag version )
Python path : ~/carla/PythonAPI/carla/dist/**carla-0.9.10-py3.6-linux-x86_64.egg** : ~/carla/PythonAPI/carla/:
Python3 version: Python 3.6.9
python version : Python 2.7.17

I have built ros_bridge successfully using catkin_make and catkin_make install.

When I run test case 
    `rostest carla_ros_bridge ros_bridge_client.test
`
It gives me below error 
```
 ... logging to /home/sandip/.ros/log/rostest-sandip-7530-14310.log
[ROSUNIT] Outputting test results to /home/sandip/.ros/test_results/carla_ros_bridge/rostest-test_ros_bridge_client.xml
Traceback (most recent call last):
  File "/home/sandip/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py", line 24, in <module>
    import carla
  File "/home/sandip/carla/PythonAPI/carla/dist/carla-0.9.10-py3.6-linux-x86_64.egg/carla/__init__.py", line 8, in <module>
    
  File "/home/sandip/carla/PythonAPI/carla/dist/carla-0.9.10-py3.6-linux-x86_64.egg/carla/libcarla.py", line 9, in <module>
  File "/home/sandip/carla/PythonAPI/carla/dist/carla-0.9.10-py3.6-linux-x86_64.egg/carla/libcarla.py", line 3, in __bootstrap__
ImportError: No module named util
```

Srichitra-S, 2022-11-10:
Same issue encountered with carla-0.9.13-py3.8-linux-x86_64.egg
Did you fix this ?

shuaishuaicha, 2024-06-29:
Have you resolved it now

SirMichcik, 2024-09-05:
same issue here