carla-simulator/ros-bridge #621: fail to build pcl_recorder

samuelseu2022, 2022-06-23:
Hello, it happens to me that pcl_recorder is failed to build with colcon build tool with ros2 galactic environment. In ros-bridge/pcl_recorder/src/PclRecorderROS2.cpp file, line 30, the callback_group namespace is not declared. I also checked latest rclcpp code. There's no callback_group namespace (https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/callback_group.hpp). Please correct me if I'm wrong.

problem code:
  sub_opt.callback_group = this->create_callback_group(rclcpp::**callback_group::**CallbackGroupType::MutuallyExclusive);

correct code:
sub_opt.callback_group = this->create_callback_group(rclcpp::CallbackGroupType::MutuallyExclusive);

youxiho1, 2022-08-11:
Ah, I can build successfully when I use ros2 foxy but I encountered into this problem too when I build with ros2 galactic environment. I've wasted lots of time in solving it.
I revised the code as you wrote here and it fixed my problem!
Thank you a lot!!!

youxiho1, 2022-08-11:
> Hello, it happens to me that pcl_recorder is failed to build with colcon build tool with ros2 galactic environment. In ros-bridge/pcl_recorder/src/PclRecorderROS2.cpp file, line 30, the callback_group namespace is not declared. I also checked latest rclcpp code. There's no callback_group namespace (https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/callback_group.hpp). Please correct me if I'm wrong.
> 
> @joel-mb seems your commit brought the changes.
> 
> ```
> //problem code:
> sub_opt.callback_group = this->create_callback_group(rclcpp::callback_group::CallbackGroupType::MutuallyExclusive);
> 
> //correct code:
> sub_opt.callback_group = this->create_callback_group(rclcpp::CallbackGroupType::MutuallyExclusive);
> ```

Ah, I can build successfully when I use ros2 foxy but I encountered into this problem too when I build with ros2 galactic environment. I've wasted lots of time in solving it.
I revised the code as you wrote here and it fixed my problem!
Thank you a lot!!!