carla-simulator/ros-bridge #627: Cannot locate rosdep definition for [derived_object_msgs]

CagriCatik, 2022-07-16:
I am currently trying to launch the carla_ros_bridge. When I run rosdep update, I get the following error message. 

```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
carla_ros_bridge: Cannot locate rosdep definition for [derived_object_msgs]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

And after  ros2 launch carla_ros_bridge carla_ros_bridge.launch.py timeout:=30
```
[bridge-1]   File "/home/cc/carla-ros-bridge/install/carla_ros_bridge/lib/python3.8/site-packages/carla_ros_bridge/traffic_participant.py", line 17, in <module>
[bridge-1]     from derived_object_msgs.msg import Object
[bridge-1] ModuleNotFoundError: No module named 'derived_object_msgs'
[ERROR] [bridge-1]: process has died [pid 33168, exit code 1, cmd '/home/cc/carla-ros-bridge/install/carla_ros_bridge/lib/carla_ros_bridge/bridge --ros-args -r __node:=carla_ros_bridge --params-file /tmp/launch_params_dvwxo1d6 --params-file /tmp/launch_params_sgiabnc6 --params-file /tmp/launch_params_tpdnwjgd --params-file /tmp/launch_params_b9ophnzq --params-file /tmp/launch_params_thxzd0xm --params-file /tmp/launch_params_huv5dsgq --params-file /tmp/launch_params__k0gsz90 --params-file /tmp/launch_params_b2fa_t50 --params-file /tmp/launch_params_d5qub8vz --params-file /tmp/launch_params_hxc2k9n4 --params-file /tmp/launch_params_w1y1bf03'].
[INFO] [launch]: process[bridge-1] was required: shutting down launched system
```

```
Distributor ID: Ubuntu
Description: Ubuntu 20.04.4 LTS
Release: 20.04
Codename: focal
ROS version: galactic
```
Does anyone have any ideas?

Light7Rain, 2022-07-23:
I have the same problem as you, and I tried to use carla_0.9.12 and carla_0.9.13, both reported the same error, no module named 'derived_object_msgs'

CagriCatik, 2022-07-23:
There is no `derived_object_msgs` for `galactic` ...

--> https://github.com/astuff/astuff_sensor_msgs/issues/82
--> http://repo.ros2.org/status_page/ros_galactic_default.html

What is your output after you run the command `rosdep install --from-paths src --ignore-src -r`?
But if you have `foxy` installed, it should work without any problem.

Light7Rain, 2022-07-29:
Thanks a lot,I changed the ros2 version to foxy,then everything went well.I'm sorry that I know nothing about your problems,but you can try this:
`mkdir -p ~/carla-ros-bridge && cd ~/carla-ros-bridge
git clone --recurse-submodules https://github.com/carla-simulator/ros-bridge.git src/ros-bridge
source /opt/ros/foxy/setup.bash
rosdep update
rosdep install --from-paths src --ignore-src -r`

Light7Rain, 2022-07-29:
`mkdir -p ~/carla-ros-bridge && cd ~/carla-ros-bridge`
`git clone --recurse-submodules https://github.com/carla-simulator/ros-bridge.git src/ros-bridge`
`source /opt/ros/foxy/setup.bash`
`rosdep update`
`rosdep install --from-paths src --ignore-src -r`

masskro0, 2023-09-23:
Had the same issue. Resolved it by running the following in my terminal:
`sudo apt install ros-foxy-derived-object-msgs 
sudo apt install ros-foxy-ackermann-msgs`