carla-simulator/ros-bridge #637: Wrong conversion from CARLA coordinate system to ROS coordinate system?

gunnxx, 2022-09-05:
Hi, I noticed that in [carla_rotation_to_RPY](https://github.com/carla-simulator/ros-bridge/blob/master/carla_common/src/carla_common/transforms.py#L81) function, **yaw and pitch are inverted**. If CARLA uses left-hand coordinate system (with z-axis up) and ROS uses right-hand coordinate system (with z-axis up), then yaw and roll should be the ones inverted. See figure below.
[image]

I do not know if there is another rule that I missed, because CARLA documentation itself is kind of inconsistent. In the [carla.Rotation](https://carla.readthedocs.io/en/latest/python_api/#carlarotation) documentation, the picture shows that the yaw is left-handed but roll and pitch are right-handed.

geyang, 2022-10-03:
Only if folks who build ROS and CARLA knew graphics, we wouldn't have to do these conversions.

UditSinghParihar, 2022-11-14:
Correct transforms should be like this in [transforms.py](https://github.com/carla-simulator/ros-bridge/blob/master/carla_common/src/carla_common/transforms.py) as mentioned by @gunnxx 

RHS (ROS) = LHS (CARLA)

Position:
```
x_ros = x_carla
y_ros = -y_carla
z_ros = z_carla

```
Rotation:
```
roll_ros = -roll_carla
pitch_ros = pitch_carla
yaw_ros = -yaw_carla
```

gunnxx, 2022-11-14:
@UditSinghParihar you also have to change the default config to spawn ego if you use that one ([here](https://github.com/carla-simulator/ros-bridge/blob/master/carla_spawn_objects/config/objects.json#L40)). You have to negate the pitch. This is very very inconvenient.

UditSinghParihar, 2022-11-14:
@gunnxx , thanks for the pointer but moving forward I have just changed the tranforms.py and the changes in the `objects.json` would only be in the right-hand coordinate transform.

swiz23, 2025-05-22:
I realize this is an old thread, but I'm in the process of trying to figure this out now as well. I've found that if I just add a simple cube to a map in Unreal, then adjust the Transform rotation in the editor (increasing the roll, pitch, yaw values), it seems to follow the picture in the [carla.Rotation](https://carla.readthedocs.io/en/latest/python_api/#carlarotation) documentation.