carla-simulator/ros-bridge #651: [0.9.13] `object_sensor` can only publish 2 object types (`car` and `pedestrain`)

heweweh, 2022-12-02:
### Problem
I'm generating training data from carla-simulator and collect samples via carla-ros-bridge.
I found that carla-ros-bridge is only giving `car` and `pedestrian`, without `bike`, `motorcycle`, `truck`, etc.
I noticed that in `carla-ros-bridge/src/carla_ros_bridge/vehicle.py:42 Vehicle.__init__()` 
```python
        self.classification = Object.CLASSIFICATION_CAR
        if 'object_type' in carla_actor.attributes:
            if carla_actor.attributes['object_type'] == 'car':
                self.classification = Object.CLASSIFICATION_CAR
            elif carla_actor.attributes['object_type'] == 'bike':
                self.classification = Object.CLASSIFICATION_BIKE
            elif carla_actor.attributes['object_type'] == 'motorcycle':
                self.classification = Object.CLASSIFICATION_MOTORCYCLE
            elif carla_actor.attributes['object_type'] == 'truck':
                self.classification = Object.CLASSIFICATION_TRUCK
            elif carla_actor.attributes['object_type'] == 'other':
                self.classification = Object.CLASSIFICATION_OTHER_VEHICLE
```
ros-bridge propose to get `object_type` from actors to determinate the actual type of `vehicle`, 
but actually the `carla_actor.attributes` will be always empty dict `{}` in my trace.

### Reproduce:
```shell
terminal 1 $ carla-simulator/CarlaUE4.sh -benchmark -fps=20
terminal 2 $ ros2 launch carla_ros_bridge carla_ros_bridge.launch.py fixed_delta_seconds:='0.1'
terminal 3 $ carla-simulator/PythonAPI/examples/generate_traffic.py -n 50 -w 15
```

berndgassmann, 2022-12-19:
Did you test the workaround from PR #641 ? At least that should work for many of the vehicles that come with CARLA 0.9.13.

heweweh, 2023-01-11:
> Did you test the workaround from PR https://github.com/carla-simulator/ros-bridge/pull/641
thanks, I did similar patch locally, I think your patch will work fine. 
the `type_id` we refered is from the blueprint library I think, you patch will do well except there are some new blueprint added to carla. but I think the root cause will still be the 'object_type' property which is readonly and from blueprint.