carla-simulator/ros-bridge #658: AttributeError: 'module' object has no attribute 'CityObjectLabel'

vimal4353, 2023-01-05:
Hi Everyone,

Environment :

OS : Ubuntu 18.04
Carla : 0.9.11(Docker image)
ROS melodic & Catkin workspace
python 2.7

I am running the carla 0.9.11 in docker container (Default map is loaded town03) & Downloaded  the carla ros bridge 0.9.11 and upload in the docker container and take a pull to local environment for running the ros in local.

I am successfully installed the rosbridge and catkin workspace by following the documentation [https://carla.readthedocs.io/en/0.9.11/ros_installation/](url) 

After installed i tried to launch the carla ros bridge by using these command :
 roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch

I am getting these following error :

Traceback (most recent call last):
  File "/home/carla/PythonAPI/docker_carla/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py", line 30, in <module>
    from carla_ros_bridge.actor_factory import ActorFactory
  File "/home/carla/PythonAPI/docker_carla/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/actor_factory.py", line 33, in <module>
    from carla_ros_bridge.marker_sensor import MarkerSensor
  File "/home/carla/PythonAPI/docker_carla/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/marker_sensor.py", line 29, in <module>
    carla.CityObjectLabel.Buildings: ColorRGBA(r=70.0/255.0, g=70.0/255.0, b=70.0/255.0, a=0.8),
AttributeError: 'module' object has no attribute 'CityObjectLabel'

Log file :
[rospy.client][INFO] 2023-01-05 16:19:51,839: init_node, name[/carla_manual_control_ego_vehicle], pid[5957]
[xmlrpc][INFO] 2023-01-05 16:19:51,840: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2023-01-05 16:19:51,840: Started XML-RPC server [http://carla:34977/]
[rospy.init][INFO] 2023-01-05 16:19:51,840: ROS Slave URI: [http://carla:34977/]
[rospy.impl.masterslave][INFO] 2023-01-05 16:19:51,840: _ready: http://carla:34977/
[rospy.registration][INFO] 2023-01-05 16:19:51,841: Registering with master node http://localhost:11311
[xmlrpc][INFO] 2023-01-05 16:19:51,841: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2023-01-05 16:19:51,940: registered with master
[rospy.rosout][INFO] 2023-01-05 16:19:51,941: initializing /rosout core topic
[rospy.rosout][INFO] 2023-01-05 16:19:51,942: connected to core topic /rosout
[rospy.simtime][INFO] 2023-01-05 16:19:51,943: initializing /clock core topic
[rospy.simtime][INFO] 2023-01-05 16:19:51,944: connected to core topic /clock
[rospy.internal][INFO] 2023-01-05 16:19:52,367: topic[/rosout] adding connection to [/rosout], count 0
[rospy.internal][INFO] 2023-01-05 16:19:52,409: topic[/carla/ego_vehicle/vehicle_control_manual_override] adding connection to [http://carla:34977/], count 0
[rospy.internal][INFO] 2023-01-05 16:19:52,438: topic[/carla/ego_vehicle/vehicle_control_manual_override] adding connection to [/carla_manual_control_ego_vehicle], count 0
[rospy.core][INFO] 2023-01-05 16:19:52,453: signal_shutdown [signal-2]
[rospy.internal][INFO] 2023-01-05 16:19:52,456: topic[/carla/ego_vehicle/vehicle_control_manual_override] removing connection to /carla_manual_control_ego_vehicle
[rospy.internal][INFO] 2023-01-05 16:19:52,456: topic[/rosout] removing connection to /rosout
[rospy.internal][INFO] 2023-01-05 16:19:52,456: topic[/carla/ego_vehicle/vehicle_control_manual_override] removing connection to http://carla:34977/
[rospy.impl.masterslave][INFO] 2023-01-05 16:19:52,457: signal-2
[rospy.core][INFO] 2023-01-05 16:19:53,095: signal_shutdown [atexit]

Is there issue in port number or version mismatch or wrong map  ?

Please help.

yaoweixiao-ux, 2023-02-13:
same question!

15138386327, 2023-03-09:
same question!