carla-simulator/ros-bridge #664: Rviz flickering when lidar rotation frequency is 10

oguzhanozfe, 2023-02-06:
When i spawned vehicle from roslaunch with Hz 10 it flickers however, when i make Hz 20 problem was gone. I tried different target fps for builds but it didn't worked. 

https://youtu.be/4lBgJriuyn0

https://youtu.be/AEz-z_cdSB8

Any help ?

oliver-schumann, 2024-03-19:
This is intended behaviour. The default server tick time is 0.05s. which means 20 Hz.
Hence, every time the server ticks and data is published, your Laserscanner only travelled half a distance. Rviz is showing you only the best it could. There are two solutions:

1. Set the lidar frequency to 20 Hz.

2. For visualizing purposes, you can set the decay time in rviz to at least 0.05s. then the points of two half scans will be visible in rviz.

oliver-schumann, 2024-09-11:
See [this Issue](https://github.com/carla-simulator/ros-bridge/issues/680) and the respective commits.
The problem was, that the partial lidar scans must be stacked if a full scan is desired :)