carla-simulator/ros-bridge #665: Low fps on carla's vehicle camera using ros bridge

camilo-0715, 2023-02-11:
I am working on a project to develop a lane detection system on Ros2 foxy using Carla ROS-Bridge to integrate it with  Carla Simulator. I am facing a major issue of low frame rate, which is affecting the performance of my application. Currently, I am receiving an average frame rate of 12-13 fps without any image processing and only 3-4 fps with image processing to detect the lanes.

I have checked the publication rate of the images topic using the ROS2 hz command and found that the publisher is working at a frequency of 10 Hz. I believe this low publication rate is the root cause of the low frame rate problem. I have tried various modifications to accelerate the publication rate, but so far have been unsuccessful.

Carla simulator runs at a decent image frecuency but when I use the ros bridge to visualize the cars camera it drops

I am seeking assistance from anyone who has knowledge and experience in accelerating the publication rate and improving the frame rate for a smoother performance on carla-to-ros-bridge

I am developing my aplication on Ubuntu Focal 20.04 on a laptop with an Intel Core i7 processor and Nvidia 3060 graphics card. I am using the latest version of Carla and Carla ROS-Bridge on ROS 2 Foxy.

zhang324-L, 2023-12-10:
I have a similar issue, have you resolved it? Looking forward to your response.

camilo-0715, 2024-01-10:
hi, no i wasnt able to fix it, it seems its a bottle neck caused by ros bridge. just changed to carla api instead of using ros