carla-simulator/ros-bridge #666: Hi Carla_ros_bridge friends

yaoweixiao-ux, 2023-02-14:
I have got a problem, when I launched `roslaunch carla_ros_bridge carla_ros_bridge.launch`
everything looks normal But the GUI is so dark, that I can't see clear car model, and ASDW can't work

hexrefugee, 2023-02-14:
Are you met this erros:
 undefined symbol: _Py_tracemalloc_config
================================================================================REQUIRED process [carla_ros_bridge-2] has died!
process has died [pid 4966, exit code 1, cmd /home/v2t-smi/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py __name:=carla_ros_bridge __log:=/home/v2t-smi/.ros/log/6e03b308-ac37-11ed-aee4-71e87aa24066/carla_ros_bridge-2.log].
log file: /home/v2t-smi/.ros/log/6e03b308-ac37-11ed-aee4-71e87aa24066/carla_ros_bridge-2*.log
Initiating shutdown!
================================================================================

hexrefugee, 2023-02-14:
Are you met this error?
` undefined symbol: _Py_tracemalloc_config
================================================================================REQUIRED process [carla_ros_bridge-2] has died!
process has died [pid 4966, exit code 1, cmd /home/v2t-smi/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py __name:=carla_ros_bridge __log:=/home/v2t-smi/.ros/log/6e03b308-ac37-11ed-aee4-71e87aa24066/carla_ros_bridge-2.log].
log file: /home/v2t-smi/.ros/log/6e03b308-ac37-11ed-aee4-71e87aa24066/carla_ros_bridge-2*.log
Initiating shutdown!
================================================================================
`

Derkai52, 2023-03-22:
Hello, it may be the cause of slow computer performance, you need to be patient or reduce the rendering quality to achieve your work

hongzhifu, 2023-04-11:
My situation is the same as yours with using "roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch", but I use "roslaunch carla_ros_bridge carla_ros_bridge.launch", I get nothing.

harrylal, 2023-11-09:
I encountered a similar issue, and it seems to be related to #469 . The solution provided there might help you with this problem. Please check it out