carla-simulator/ros-bridge #675: client can not get actors from carla server when using ros2 carla bridge

TurtleZhong, 2023-04-27:
Hi 
Follwing this [ROS Bridge package](https://carla.readthedocs.io/projects/ros-bridge/en/latest/ros_installation_ros2/),  , I use the ros2 version and I can get ros2 topics from CARLA.

but when I wanna use carla python API to get the actors, I got some error.

Acutually I do this steps

`./CarlaUE4.sh`
`ros2 launch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch.py`
`python3 control_corla.py`
below is the code of control_carla.py

```python
#!/usr/bin/python3
# -*- coding:utf-8 -*-

import glob
import os
import sys
try:
    sys.path.append(glob.glob('**/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
    sys.path.append("./")
except IndexError:
    pass

# ==============================================================================
# -- imports -------------------------------------------------------------------
# ==============================================================================
import carla
import numpy as np
import random
import rclpy
from rclpy.node import Node
from ros_g29_force_feedback.msg import ForceFeedback

class CarlaFFNode(Node):
    def __init__(self, client):
        super().__init__("carla_ff_node")
        self.publisher = self.create_publisher(ForceFeedback,"/ff_target", 10)
        self.timer = self.create_timer(0.1, self.timer_cb)
        self.client = client
        self.world = client.get_world()
        print(self.world.__str__())
        self.actor = self.get_hero()

    def get_hero(self):
        if(self.world.get_actors() is not None:
                print('The world has no actors')
                print(self.world.get_actors().__len__())
                print(self.world.get_actors().__str__())
        for actor in self.world.get_actors():
            print(actor)
            print(actor.attributes)
            if actor.attributes.get("role_name") in ["hero", "ego_vehicle"]:
                return actor
            
    def timer_cb(self):
        out_msg = ForceFeedback()
        out_msg.header.stamp = self.get_clock().now().to_msg()
        
        steering_angle = self.actor.get_control().steer
        out_msg.position = steering_angle
        out_msg.torque = 0.8
        self.publisher.publish(out_msg)

def main(args=None):

    rclpy.init(args=args)
    client = carla.Client("127.0.0.1", 2000)
    client.set_timeout(2.0)

    carla_ff_node = CarlaFFNode(client)
    rclpy.spin(carla_ff_node)
    carla_ff_node.destroy_node()
    rclpy.shutdown()

if __name__ == "__main__":
    main()
```
Then I got
[image]

But if I only run carla server and use the carla example mannual control
`./CarlaUE4.sh`
in example dir
`python3 mannual_control.py`
`python3 control_corla.py`

I got this , I can get the actors in carla client.
[image]

qzhq747, 2024-10-17:
I also face the same issue. I use ros-noetic carla bridge.

yumemide, 2025-03-03:
call world.wait_for_tick() before first time to get_actors()