carla-simulator/ros-bridge #679: Sensor topics not registering

MarioAlmelaUMH, 2023-05-16:
Hello, I'm currently working with Carla and carla-ros-bridge, and I need to trace sensor data from the scene into a .csv file. I have already made a script that captures that data directly using Carla's API, but it is in my interest to capture the same data using carla-ros-bridge's topics and the rosbag module. 

I have a custom config file where I can modify the sensor kit of the ego vehicle, and I read that config file in order to spawn all the sensors using Carla's API. Right now, I'm trying a sensor kit with 4 cameras, 3 lidars, 1 GNSS and 1 IMU. When I run the carla-ros-bridge, I see the bridge captures all the sensors I spawned:

[image]

However, the ros-bridge is not creating proper topics to monitor these sensors, as seen in:

[image]

Is there a way I can automatically generate topics with format /carla/<vehicle_role>/<sensor_name>/(image, pointcloud, etc.) for each sensor individually?

I've seen that using the carla_ros_bridge_with_example_ego_vehicle.launch.py, the carla-ros-bridge itself launches an ego vehicle with a lot more topics, as seen in:
 
[image]

Is there a way that I can get this amount of sensor topics but using my own custom ego_vehicle spawn script?

Thank you in advance.

MarioAlmelaUMH, 2023-05-23:
Anyone has any insight on this? 

Thank you in advance.

celik02, 2024-06-10:
I am having exactly the same problem. Does anyone have any suggestion? Thank you.