carla-simulator/ros-bridge #680: Incomplete lidar points cloud when sensor_tick doesn't match simulation frequency

Paul-Nico, 2023-06-06:
Hello, I noticed a bug on the lidar sensor :  if simulation frequency is higher than lidar frequency, the published lidar point cloud will only contains data produced on the last simulation tick, and will then be incomplete. 
I have written a simple fix for that, which stack the produced lidar data when the simulation ticks and publish the whole packet when the sensors ticks. I would be happy to do a pull request if possible.
Best regards,
Paul

oliver-schumann, 2024-09-11:
Thank you for this branch! I stumbled over this problem some months ago and you just made my day!
I also ported it to the semantic lidar. Works like a charm :)

Paul-Nico, 2024-09-12:
> Thank you for this branch! I stumbled over this problem some months ago and you just made my day! I also ported it to the semantic lidar. Works like a charm :)

Glad this helped :)