carla-simulator/ros-bridge #683: Missing packages in debian installation

murthybalaji0725, 2023-06-13:
**Ubuntu 18.04
Carla **0.9.14**
carla-ros-bridge 0.9.10-1**

I performed a debian installation of the rosbridge and I am trying to run 
`roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch`

However, on running the pygame window in black. I can change from manual to auto using the keys but I cannot control the vehicle. The vehicle spawns correctly within unreal engine. 
I think that are no sensors spawned and i checked in the /opt/carla-ros-bridge and there are many packages missing. Especially the carla_spawn_objects. 

How can I add additional packages when I have a debian installation. I tried the source repository method too, but it gives me an error saying 

```
ERROR: the following rosdeps failed to install
  pip: command [sudo -H pip2 install -U transforms3d] failed
  pip: Failed to detect successful installation of [transforms3d]
```
I need help to add the rest of packages that are shown in the repository.

peter50107, 2023-06-20:
maybe you should use the new version carla_ros_bridge.

murthybalaji0725, 2023-06-20:
The new version of carla_ros_bridge gives me an error with transforms3d package. Which is why I am trying to do debian, because it mostly works. However in debian it is available till 0.9.10-1 which doesnt have the additional packages