carla-simulator/ros-bridge #691: fatal error: tf2_eigen/tf2_eigen.h: No such file or directory-- colcon build error while installing carla-ros bridge

AJK97, 2023-07-11:
hey community, I am a student and am trying to install carla ros bridge for the first time, but I keep encountering this error during the installation process. Anybody has experienced this before? Any help greatly appreciated!!!
ROS2 Humble, Ubuntu 22.04, Carla 9.14

[image]

gezp, 2023-07-12:
i fixed this problem by updating `ros-bridge/pcl_recorder/CMakeLists.txt`, you should add `tf2_eigen` to `ament_target_dependencies()`.

raw:
https://github.com/carla-simulator/ros-bridge/blob/e9063d97ff5a724f76adbb1b852dc71da1dcfeec/pcl_recorder/CMakeLists.txt#L68-L69

correct:
```cmake
  ament_target_dependencies(${PROJECT_NAME}_node rclcpp sensor_msgs
                            pcl_conversions tf2 tf2_ros tf2_eigen)
```

AJK97, 2023-07-12:
hi, i tried it out and got this 
[image]

gezp, 2023-07-13:
hi, i'm also adapting `ros bridge` of carla 0.9.14 for ROS humble on ubuntu22.04, you can try a fork of this repo [carla_ros](https://github.com/gezp/carla_ros/tree/humble-carla-0.9.14) , which works well for me.

AJK97, 2023-07-14:
Hi, thanks for this! I will give it a try. one thing is that for me carla 9.14 is running in docker with ubuntu 20.04. ROS humble is installed on the workstation that has 22.04. So this should still be fine right?

And should I follow the README file instructions AFTER the ros carla bridge installation procedure as per the official documentation (https://carla.readthedocs.io/projects/ros-bridge/en/latest/ros_installation_ros2/ )? or INSTEAD?

gezp, 2023-07-14:
> one thing is that for me carla 9.14 is running in docker with ubuntu 20.04 and ROS2 humble is installed on the workstation that has ubuntu 22.04. So this should still be fine right?

i think it's ok, i do  the same way.

> Im quite an amateur so it would be helpful if you could provide some steps that should follow?

only need follow the README in [carla_ros](https://github.com/gezp/carla_ros/tree/humble-carla-0.9.14).

AJK97, 2023-07-17:
I tried it but I get tje error "WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
given path 'src' does not exist". Any idea?

[image]

AJK97, 2023-07-17:
HI, thanks for the explanation! I believe I got few warnings, i believe its okay to ignore these warnings?
or maybe supress them with using PYTHONWARNINGS variable? 

[image]

gezp, 2023-07-17:
> Now finally to run the bridge, how is the process different form the official documentation since CARLA is running in a docker here?

yes, you can follow official documentation to launch carla ros bridge. by the way, you should add argument `host`  with your carla server ip.

AJK97, 2023-07-19:
Hi, just follow the read me file in the git repo mentioned above. Before that build a workspace first according to https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Creating-A-Workspace/Creating-A-Workspace.html in which you will clone and then build according to the above exchange.

AJK97, 2023-07-19:
@gezp Thanks once again for your help!!! I am closing this issue now :)

closed 2023-07-19 by AJK97

pradhanshrijal, 2023-12-28:
Hey, this works, did you try to make a pull request to the main repo?