carla-simulator/ros-bridge #692: Setting “sensor_tick” in carla_spawn_objects/config/objects.json doesn't work

suyuzhu, 2023-07-12:
Why do I set  **"fixed_delta_seconds"**  in _ros-bridge/carla_ros_bridge/launch/carla_ros_bridge_with_example_ego_vehicle.launch_ to 0.01,and set the camera's **"sensor_tick"** in _carla_spawn_objects/config/objects.json_ to 0.02, but the rostopic hz /carla/ego_vehicle/rgb_front/image is only 10~20, and the camera frequency is different every time I roslaunch  carla_ros_bridge_with_example_ego_vehicle.launch

This is the  _carla_ros_bridge_with_example_ego_vehicle.launch_ I set:
```
<!-- -->
<launch>
  <!-- CARLA connection -->
  <arg name='host' default='localhost'/>
  <arg name='port' default='2000'/>
  <arg name='timeout' default='10'/>

  <!-- Ego vehicle -->
  <arg name='role_name' default='ego_vehicle'/>
  <arg name="vehicle_filter" default='vehicle.*'/>
  <arg name="spawn_point" default=""/><!-- use comma separated format "x,y,z,roll,pitch,yaw" -->

  <!-- Map to load on startup (either a predefined CARLA town (e.g. 'Town01'), or a OpenDRIVE map file) -->
  <arg name="town" default='Town03'/>

  <!-- Enable/disable passive mode -->
  <arg name='passive' default='true'/>

  <!-- Synchronous mode-->
  <arg name='synchronous_mode_wait_for_vehicle_control_command' default='False'/>
  <arg name='fixed_delta_seconds' default='0.01'/>

  <include file="$(find carla_ros_bridge)/launch/carla_ros_bridge.launch">
    <arg name='host' value='$(arg host)'/>
    <arg name='port' value='$(arg port)'/>
    <arg name='town' value='$(arg town)'/>
    <arg name='timeout' value='$(arg timeout)'/>
    <arg name='passive' value='$(arg passive)'/>
    <arg name='synchronous_mode_wait_for_vehicle_control_command' value='$(arg synchronous_mode_wait_for_vehicle_control_command)'/>
    <arg name='fixed_delta_seconds' value='$(arg fixed_delta_seconds)'/>
  </include>

  <!-- the ego vehicle, that will be controlled by an agent (e.g. carla_ad_agent) -->
  <include file="$(find carla_spawn_objects)/launch/carla_example_ego_vehicle.launch">
    <arg name="objects_definition_file" value='$(find carla_spawn_objects)/config/objects.json'/>
    <arg name='role_name' value='$(arg role_name)'/>
    <arg name="spawn_point_ego_vehicle" value="$(arg spawn_point)"/>
    <arg name="spawn_sensors_only" value="false"/>
  </include>

  <include file="$(find carla_manual_control)/launch/carla_manual_control.launch">
    <arg name='role_name' value='$(arg role_name)'/>
  </include>

</launch>
```

And the _objects.json_ camera part is :
```
 "sensors": 
            [
                {
                    "type": "sensor.camera.rgb",
                    "id": "rgb_front",
                    "spawn_point": {"x": 2.0, "y": 0.0, "z": 2.0, "roll": 0.0, "pitch": 0.0, "yaw": 0.0},
                    "image_size_x": 800,
                    "image_size_y": 600,
                    "fov": 90.0,
                    "sensor_tick":0.02
                },
```
When I run the bridge

> roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch

It can show the **fixed_delta_seconds** is right
[image]

and check the frequence of camera

> rostopic hz /carla/ego_vehicle/rgb_front/image

It appears like this:
[image]

I really don't know what to do to get higher frequence of the sensor.
Can anyone give some advice? 
Thanks!

Billthekidz, 2023-09-19:
I recently encounter another issue with timing: I want my GNSS sensor to publish at 1 Hz, and it does, but the flow of time inside the simulation is slower than the real time.
But all I have to say for your issue is that the synchronous mode + fixed_delta_seconds setting could have a huge impact on your real-time performance, not just how smooth the simulation is, but also how accurate your sensor frequency is.

So study the doc about these topic and wish the best.

goodgoodliving, 2023-10-27:
> 为什么我将 _ros-bridge/carla_ros_bridge/launch/carla_ros_bridge_with_example_ego_vehicle.launch_ 中的“**fixed_delta_seconds**”设置为 0.01，而在 _carla_spawn_objects/config/objects.json_ 中将相机的**“sensor_tick”**设置为 0.02，但 rostopic hz /carla/ego_vehicle/rgb_front/image 只有 10~20，而且每次 roslaunch carla_ros_bridge_with_example_ego_vehicle.launch 时相机频率都不一样
> 
> 这是我设置_的carla_ros_bridge_with_example_ego_vehicle.launch_：
> 
> ```
> 
> <launch>
>   
>   <arg name='host' default='localhost'/>
>   <arg name='port' default='2000'/>
>   <arg name='timeout' default='10'/>
> 
>   
>   <arg name='role_name' default='ego_vehicle'/>
>   <arg name="vehicle_filter" default='vehicle.*'/>
>   <arg name="spawn_point" default=""/>
> 
>   
>   <arg name="town" default='Town03'/>
> 
>   
>   <arg name='passive' default='true'/>
> 
>   
>   <arg name='synchronous_mode_wait_for_vehicle_control_command' default='False'/>
>   <arg name='fixed_delta_seconds' default='0.01'/>
> 
> 
>   <include file="$(find carla_ros_bridge)/launch/carla_ros_bridge.launch">
>     <arg name='host' value='$(arg host)'/>
>     <arg name='port' value='$(arg port)'/>
>     <arg name='town' value='$(arg town)'/>
>     <arg name='timeout' value='$(arg timeout)'/>
>     <arg name='passive' value='$(arg passive)'/>
>     <arg name='synchronous_mode_wait_for_vehicle_control_command' value='$(arg synchronous_mode_wait_for_vehicle_control_command)'/>
>     <arg name='fixed_delta_seconds' value='$(arg fixed_delta_seconds)'/>
>   </include>
> 
>   
>   <include file="$(find carla_spawn_objects)/launch/carla_example_ego_vehicle.launch">
>     <arg name="objects_definition_file" value='$(find carla_spawn_objects)/config/objects.json'/>
>     <arg name='role_name' value='$(arg role_name)'/>
>     <arg name="spawn_point_ego_vehicle" value="$(arg spawn_point)"/>
>     <arg name="spawn_sensors_only" value="false"/>
>   </include>
> 
>   <include file="$(find carla_manual_control)/launch/carla_manual_control.launch">
>     <arg name='role_name' value='$(arg role_name)'/>
>   </include>
> 
> </launch>
> ```
> 
> _objects.json_相机部分是：
> 
> ```
>  "sensors": 
>             [
>                 {
>                     "type": "sensor.camera.rgb",
>                     "id": "rgb_front",
>                     "spawn_point": {"x": 2.0, "y": 0.0, "z": 2.0, "roll": 0.0, "pitch": 0.0, "yaw": 0.0},
>                     "image_size_x": 800,
>                     "image_size_y": 600,
>                     "fov": 90.0,
>                     "sensor_tick":0.02
>                 },
> ```
> 
> 当我跑桥时
> 
> > roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch
> 
> 它可以显示**fixed_delta_seconds**是正确的 [image]
> 
> 并检查相机的频率
> 
> > Rostopic hz /carla/ego_vehicle/rgb_front/Image
> 
> 它看起来像这样： [image]
> 
> 我真的不知道该怎么做才能获得更高的传感器频率。谁能给出一些建议？ 谢谢！

> 为什么我将 _ros-bridge/carla_ros_bridge/launch/carla_ros_bridge_with_example_ego_vehicle.launch_ 中的“**fixed_delta_seconds**”设置为 0.01，而在 _carla_spawn_objects/config/objects.json_ 中将相机的**“sensor_tick”**设置为 0.02，但 rostopic hz /carla/ego_vehicle/rgb_front/image 只有 10~20，而且每次 roslaunch carla_ros_bridge_with_example_ego_vehicle.launch 时相机频率都不一样
> 
> 这是我设置_的carla_ros_bridge_with_example_ego_vehicle.launch_：
> 
> ```
> 
> <launch>
>   
>   <arg name='host' default='localhost'/>
>   <arg name='port' default='2000'/>
>   <arg name='timeout' default='10'/>
> 
>   
>   <arg name='role_name' default='ego_vehicle'/>
>   <arg name="vehicle_filter" default='vehicle.*'/>
>   <arg name="spawn_point" default=""/>
> 
>   
>   <arg name="town" default='Town03'/>
> 
>   
>   <arg name='passive' default='true'/>
> 
>   
>   <arg name='synchronous_mode_wait_for_vehicle_control_command' default='False'/>
>   <arg name='fixed_delta_seconds' default='0.01'/>
> 
> 
>   <include file="$(find carla_ros_bridge)/launch/carla_ros_bridge.launch">
>     <arg name='host' value='$(arg host)'/>
>     <arg name='port' value='$(arg port)'/>
>     <arg name='town' value='$(arg town)'/>
>     <arg name='timeout' value='$(arg timeout)'/>
>     <arg name='passive' value='$(arg passive)'/>
>     <arg name='synchronous_mode_wait_for_vehicle_control_command' value='$(arg synchronous_mode_wait_for_vehicle_control_command)'/>
>     <arg name='fixed_delta_seconds' value='$(arg fixed_delta_seconds)'/>
>   </include>
> 
>   
>   <include file="$(find carla_spawn_objects)/launch/carla_example_ego_vehicle.launch">
>     <arg name="objects_definition_file" value='$(find carla_spawn_objects)/config/objects.json'/>
>     <arg name='role_name' value='$(arg role_name)'/>
>     <arg name="spawn_point_ego_vehicle" value="$(arg spawn_point)"/>
>     <arg name="spawn_sensors_only" value="false"/>
>   </include>
> 
>   <include file="$(find carla_manual_control)/launch/carla_manual_control.launch">
>     <arg name='role_name' value='$(arg role_name)'/>
>   </include>
> 
> </launch>
> ```
> 
> _objects.json_相机部分是：
> 
> ```
>  "sensors": 
>             [
>                 {
>                     "type": "sensor.camera.rgb",
>                     "id": "rgb_front",
>                     "spawn_point": {"x": 2.0, "y": 0.0, "z": 2.0, "roll": 0.0, "pitch": 0.0, "yaw": 0.0},
>                     "image_size_x": 800,
>                     "image_size_y": 600,
>                     "fov": 90.0,
>                     "sensor_tick":0.02
>                 },
> ```
> 
> 当我跑桥时
> 
> > roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch
> 
> 它可以显示**fixed_delta_seconds**是正确的 [image]
> 
> 并检查相机的频率
> 
> > Rostopic hz /carla/ego_vehicle/rgb_front/Image
> 
> 它看起来像这样： [image]
> 
> 我真的不知道该怎么做才能获得更高的传感器频率。谁能给出一些建议？ 谢谢！

Have you resolved it? I'm encountering the same issue now.