carla-simulator/ros-bridge #693: exit code -11

stepan-antonyuk, 2023-07-13:
Hello, I have issue launching a simulation. It worked before, but now it gives exit code -11 every time I try to run it. I am using ros2

Here is what I do:
santonyuk@justingpu:~/Projects/navigator$ ros2 launch carla.launch.py

And this is what I get
[INFO] [launch]: All log files can be found below /home/santonyuk/.ros/log/2023-07-10-15-18-26-804056-justingpu-769698
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [carla_spawn_objects-2]: process started with pid [769701]
[INFO] [bridge-1]: process started with pid [769699]
[INFO] [liaison_node-3]: process started with pid [769703]
[INFO] [rosbridge_websocket-4]: process started with pid [769705]
[INFO] [robot_state_publisher-5]: process started with pid [769707]
[INFO] [lidar_processing_node-6]: process started with pid [769709]
[INFO] [gnss_processing_node-7]: process started with pid [769711]
[robot_state_publisher-5] [WARN] [1689020307.472845637] [robot_state_publisher]: No robot_description parameter, but command-line argument available. Assuming argument is name of URDF file. This backwards compatibility fallback will be removed in the future.
[robot_state_publisher-5] [INFO] [1689020307.476401347] [robot_state_publisher]: got segment base_link
[robot_state_publisher-5] [INFO] [1689020307.476417747] [robot_state_publisher]: got segment hero
[robot_state_publisher-5] [INFO] [1689020307.476422857] [robot_state_publisher]: got segment model_link
[rosbridge_websocket-4] [INFO] [1689020307.949480970] [rosbridge_websocket]: Rosbridge WebSocket server started on port 9091
[ERROR] [bridge-1]: process has died [pid 769699, exit code -11, cmd '/home/share/carla-ros-bridge/install/carla_ros_bridge/lib/carla_ros_bridge/bridge --ros-args -r __node:=carla_ros_bridge --params-file /tmp/launch_params_i25nf6wy --params-file /tmp/launch_params_1bwhdl5f --params-file /tmp/launch_params_0jc72yml --params-file /tmp/launch_params_cjpgfi_n --params-file /tmp/launch_params_okcgd9hc --params-file /tmp/launch_params_xwycbwxz --params-file /tmp/launch_params_eskyn9a2 --params-file /tmp/launch_params_zqz0h_h3 -r /carla/hero/lidar:=/lidar/fused -r /carla/hero/rgb_center/image:=/cameras/camera0'].
[ERROR] [liaison_node-3]: process has died [pid 769703, exit code -11, cmd '/home/santonyuk/Projects/navigator/install/carla_interface/lib/carla_interface/liaison_node --ros-args'].

WZWAsuna, 2023-07-26:
I have the same problem

oliver-schumann, 2023-11-17:
Me, too. It works only with foxy in my case.