carla-simulator/ros-bridge #698: Shutting down node using Ctrl+C, doesnt completely destroy all the spawned objects

AJK97, 2023-08-09:
Hi,

Once I destroy the carla_spawn_object node that has spawned an ego vehicle, using Ctrl+C or killall, the node dissapears from the output of "ros2 node list" but nevertheless, the car and some of the sensors are not completely destroyed since I can still visualize the data in RVIZ.
Any leads?

[image]

Billthekidz, 2023-09-19:
It looks like the shutdown process is throwing an exception so the cleanup wasn't successful.

1157457050, 2024-12-13:
I also encountered this problem. Have you solved it?

Billthekidz, 2024-12-19:
I suggest making sure ros-bridge is not shut down before any other ros-bridge nodes.
I always have it when I kill carla and the ros-bridge together too quickly, but it was okay for me because carla will be stopped and restarted anyways.