carla-simulator/ros-bridge #708: ROS2      -----> 😿[ERROR] [bridge-1]: process has died😿 <-----

pro4aesthetic, 2023-11-09:
Hi, has anyone managed to solve the problem when running the command:  
```
ros2 launch carla_ros_bridge carla_ros_bridge.launch.py
```
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2023-11-09-10-07-53-605908-user-System-Product-Name-38687
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [bridge-1]: process started with pid [38688]
[ERROR] [bridge-1]: process has died [pid 38688, exit code -11, cmd '/home/user/carla-ros-bridge/install/carla_ros_bridge/lib/carla_ros_bridge/bridge --ros-args -r __node:=carla_ros_bridge --params-file /tmp/launch_params_j_fauzyq --params-file /tmp/launch_params_njuzb6_l --params-file /tmp/launch_params_upm3qmoa --params-file /tmp/launch_params_dql7207k --params-file /tmp/launch_params_nyr0vrrv --params-file /tmp/launch_params_h3vmjk12 --params-file /tmp/launch_params_xsyxnrin --params-file /tmp/launch_params_9z6ylm2y --params-file /tmp/launch_params_m3vn8d5t --params-file /tmp/launch_params_b6r_fo84 --params-file /tmp/launch_params_1py14mri'].
[INFO] [launch]: process[bridge-1] was required: shutting down launched system  

I did it according to the [instructions](https://carla.readthedocs.io/projects/ros-bridge/en/latest/ros_installation_ros2/), but something went wrong

oliver-schumann, 2023-11-14:
I have the same problem when trying to build the bridge in the docker with iron.
Did you solve the issue ye?
However, I achieved to communicate with ros2 iron nodes and the bridge in foxy using cyclone dds.
Of course, it would be way better to let the bridge run with newer ros distributions as well.

pro4aesthetic, 2023-11-15:
I stopped steaming and run through dockerfile  
```
ARG CARLA_VERSION
ARG ROS_DISTRO

FROM carlasim/carla:$CARLA_VERSION as carla

FROM ros:$ROS_DISTRO-ros-base

ARG CARLA_VERSION
ARG ROS_DISTRO

ENV CARLA_VERSION=$CARLA_VERSION
ENV DEBIAN_FRONTEND=noninteractive

RUN apt-get update && apt-get install -y \
    ros-$ROS_DISTRO-ros-base \
    ros-$ROS_DISTRO-rmw-cyclonedds-cpp \
    && rm -rf /var/lib/apt/lists/*

ARG USER=carla-ros-bridge
ARG UID=1000
ARG GID=1000
ARG PW=user

RUN useradd -m ${USER} --uid=${UID} && echo "${USER}:${PW}" | chpasswd

RUN mkdir -p /opt/${USER}/src
WORKDIR /opt/${USER}
RUN cd /opt/${USER} && chown -R ${UID}:${GID} ./
COPY --from=carla /home/carla/PythonAPI /opt/carla/PythonAPI

COPY requirements.txt /opt/${USER}
COPY install_dependencies.sh /opt/${USER}
RUN /bin/bash -c 'source /opt/ros/$ROS_DISTRO/setup.bash; \
    bash /opt/${USER}/install_dependencies.sh; \
    if [ "$CARLA_VERSION" = "0.9.10" ] || [ "$CARLA_VERSION" = "0.9.10.1" ]; then wget https://carla-releases.s3.eu-west-3.amazonaws.com/Backup/carla-0.9.10-py2.7-linux-x86_64.egg -P /opt/carla/PythonAPI/carla/dist; fi; \
    echo "export PYTHONPATH=\$PYTHONPATH:/opt/carla/PythonAPI/carla/dist/$(ls /opt/carla/PythonAPI/carla/dist | grep py$ROS_PYTHON_VERSION.)" >> /opt/carla/setup.bash; \
    echo "export PYTHONPATH=\$PYTHONPATH:/opt/carla/PythonAPI/carla" >> /opt/carla/setup.bash'

COPY . /opt/${USER}/src/
RUN /bin/bash -c 'source /opt/ros/$ROS_DISTRO/setup.bash; \
    if [ "$ROS_VERSION" == "2" ]; then colcon build; else catkin_make install; fi'

# replace entrypoint
COPY ./docker/content/ros_entrypoint.sh /
USER ${UID}:${GID}
```
run.sh  
```
#!/bin/sh

usage() { echo "Usage: $0 [-t <tag>] [-i <image>] [-c <command>]" 1>&2; exit 1; }

# Defaults
DOCKER_IMAGE_NAME="carla-ros-bridge"
TAG="foxy"
COMMAND=""
ARCH=`uname -m`

while getopts ":ht:i:c:" opt; do
  case $opt in
    h)
      usage
      exit
      ;;
    t)
      TAG=$OPTARG
      ;;
    i)
      DOCKER_IMAGE_NAME=$OPTARG
      ;;
    c)
      COMMAND=$OPTARG
      ;;
    \?)
      echo "Invalid option: -$OPTARG" >&2
      exit 1
      ;;
    :)
      echo "Option -$OPTARG requires an argument." >&2
      exit 1
      ;;
  esac
done
shift $((OPTIND-1))

echo "Using $DOCKER_IMAGE_NAME:$TAG"

if [ "$ARCH"=="x86_64" ]; then
    ARGS="--ipc host --gpus all -e NVIDIA_DRIVER_CAPABILITIES=all"
elif [ "$ARCH"=="aarch64" ]; then
    ARGS="--runtime nvidia"
else
    echo "Arch ${ARCH} not supported"
    exit
fi

docker run \
    -it --rm \
    $ARGS \
    --net=host \
    --env="DISPLAY=$DISPLAY" \
    --env="QT_X11_NO_MITSHM=1" \
    --privileged \
    -v /tmp/.X11-unix:/tmp/.X11-unix:rw \
    "$DOCKER_IMAGE_NAME:$TAG" "$@" \
    $COMMAND
```

closed 2023-11-15 by pro4aesthetic