carla-simulator/ros-bridge #711: Linker error librviz_carla_plugin.so: undefined symbol: fromMsg

FelipeGdM, 2023-12-04:
After compiling the code from master and running the command

```bash
rviz2 -d src/ros-bridge/carla_ad_demo/config/carla_ad_demo_ros2.rviz
```

The rviz2 windows briefly appears in the screen and then the error message is shown in terminal

```bash
rviz2: symbol lookup error: /home/felipe/Documents/Github/carla_ws/install/rviz_carla_plugin/lib/librviz_carla_plugin.so: undefined symbol: _ZN3tf27fromMsgERKN13geometry_msgs3msg11Quaternion_ISaIvEEERNS_10QuaternionE
```

I believe it is a problem related with the geometry2 library and not with carla ros bridge itself.

A temporary workaround is adding a bogus declaration of the function `fromMsg`

```cpp
namespace tf2{
  void fromMsg(const geometry_msgs::msg::Quaternion & in, tf2::Quaternion & out){
    (void)(in);
    (void)(out);
  };
}
```

With this declaration, the plugin is loaded by rviz2 as intended

junge1a520, 2023-12-20:
Please tell me where to add this code,thanks.

FelipeGdM, 2023-12-20:
@junge1a520 I suggest you to try the version in my PR, it uses a more effective solution https://github.com/carla-simulator/ros-bridge/pull/713