carla-simulator/ros-bridge #712: My ROS system is ROS2 foxy, but there was an error compiling ROS bridge on Ubuntu 20. I don't know how to solve it

xoxclouds, 2023-12-05:
Starting >>> carla_msgs
Starting >>> ros_compatibility
Starting >>> carla_common                                 
Starting >>> carla_ros_scenario_runner_types              
Starting >>> carla_waypoint_types
Starting >>> carla_twist_to_control                       
Starting >>> rqt_carla_control                            
--- stderr: carla_msgs                                          
CMake Error at /opt/ros/foxy/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:60 (message):
  execute_process(/home/yrj/anaconda3/bin/python3 -m rosidl_adapter
  --package-name carla_msgs --arguments-file
  /home/yrj/carla-ros2-bridge/build/carla_msgs/rosidl_adapter__arguments__carla_msgs.json
  --output-dir
  /home/yrj/carla-ros2-bridge/build/carla_msgs/rosidl_adapter/carla_msgs
  --output-file
  /home/yrj/carla-ros2-bridge/build/carla_msgs/rosidl_adapter/carla_msgs.idls)
  returned error code 1:

  AttributeError processing template 'msg.idl.em'

  Traceback (most recent call last):

    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/resource/__init__.py", line 51, in evaluate_template
      em.BUFFERED_OPT: True,

  AttributeError: module 'em' has no attribute 'BUFFERED_OPT'

  

  During handling of the above exception, another exception occurred:

  

  Traceback (most recent call last):

    File "/home/yrj/anaconda3/lib/python3.7/runpy.py", line 193, in _run_module_as_main
      "__main__", mod_spec)
    File "/home/yrj/anaconda3/lib/python3.7/runpy.py", line 85, in _run_code
      exec(code, run_globals)
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/__main__.py", line 19, in <module>
      sys.exit(main())
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/main.py", line 55, in main
      pathlib.Path(relative_path), output_dir)
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/__init__.py", line 20, in convert_to_idl
      package_dir, package_name, interface_file, output_dir / 'msg')
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/msg/__init__.py", line 39, in convert_msg_to_idl
      expand_template('msg.idl.em', data, output_file, encoding='iso-8859-1')
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/resource/__init__.py", line 23, in expand_template
      content = evaluate_template(template_name, data)
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/resource/__init__.py", line 69, in evaluate_template
      _interpreter.shutdown()

  AttributeError: 'NoneType' object has no attribute 'shutdown'

Call Stack (most recent call first):
  /opt/ros/foxy/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:130 (rosidl_adapt_interfaces)
  CMakeLists.txt:79 (rosidl_generate_interfaces)

---
Failed   <<< carla_msgs [1.27s, exited with code 1]
Aborted  <<< carla_ros_scenario_runner_types [1.05s]
Aborted  <<< carla_waypoint_types [0.98s]
Aborted  <<< carla_common [1.27s]                                           
Aborted  <<< carla_twist_to_control [1.07s]
Aborted  <<< ros_compatibility [1.36s]
Aborted  <<< rqt_carla_control [1.00s]

Summary: 0 packages finished [2.82s]
  1 package failed: carla_msgs
  6 packages aborted: carla_common carla_ros_scenario_runner_types carla_twist_to_control carla_waypoint_types ros_compatibility rqt_carla_control
  3 packages had stderr output: carla_msgs carla_ros_scenario_runner_types carla_waypoint_types
  12 packages not processed

FelipeGdM, 2023-12-08:
Have you installed all dependencies? This error may be caused by a missing dependency

You may try 

```
rosdep install --from-paths src --ignore-src -r -y
```

Then try to compile the project again with

```
colcon build --cmake-clean-first
```

The flag `--cmake-clean-first` is needed because some intermediary object files may be broken due to missing dependencies, so it's better to attempt a fresh start

zhangjiatao, 2023-12-14:
I also encountered this issue, and the compilation error occurred on Python 3.11. I was able to resolve it by recompiling on Python 3.8.

xoxclouds, 2023-12-19:
Thank you for your feedback. I have already resolved this issue on my end.

FelipeGdM, 2023-12-19:
@xoxclouds can you post the solution? It may be useful for other people in the future

xoxclouds, 2024-01-02:
@FelipeGdM My ubuntu system crashed. I reinstalled ubuntu and compiled again and there was no problem. I may not be able to provide much help, I'm very sorry.

closed 2024-01-18 by xoxclouds

is-whale, 2024-02-12:
> 是否安装了所有依赖项？此错误可能是由缺少依赖项引起的
> 
> 你可以试试
> 
> ```
> rosdep install --from-paths src --ignore-src -r -y
> ```
> 
> 然后尝试再次编译项目
> 
> ```
> colcon build --cmake-clean-first
> ```
> 
> 该标志是必需的，因为某些中间对象文件可能会由于缺少依赖项而损坏，因此最好尝试重新开始`--cmake-clean-first`

oh ! I love U so much!!!

JiBingdong, 2024-03-21:
> @xoxclouds can you post the solution? It may be useful for other people in the future

I also encountered this issue, do you resolved this issue?
PS: i try your answer in the top, but it does not work on my system.

JiBingdong, 2024-03-21:
> > @xoxclouds can you post the solution? It may be useful for other people in the future
> 
> I also encountered this issue, do you resolved this issue? PS: i try your answer in the top, but it does not work on my system.
I have already resolved this issue on my end.
```

``` pip install empy==3.3.4

cyKkk0, 2025-01-16:
> > > @xoxclouds can you post the solution? It may be useful for other people in the future
> > 
> > 
> > I also encountered this issue, do you resolved this issue? PS: i try your answer in the top, but it does not work on my system.
> > I have already resolved this issue on my end.
> 
> ```
> 
> ``` pip install empy==3.3.4
> ```

this worked for me,thank you. I used empy v4.2 and that's incorrect

CraigBuilds, 2025-03-17:
```pip install empy==3.3.4```

This fixed it for me. I'm using ros2 humble with Windows 11.