carla-simulator/ros-bridge #737: colcon error casued by the lack of dependence `tf2_eigen`

GoesM, 2024-07-12:
## description
when I `colcon build`  ros-bridge within ROS2-humble, I faced to the following errors:
```sh
--- stderr: pcl_recorder                               
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
In file included from /home/***/ros-bridge/pcl_recorder/src/mainROS2.cpp:7:
/home***/ros-bridge/pcl_recorder/include/PclRecorderROS2.h:13:10: fatal error: 'tf2_eigen/tf2_eigen.h' file not found
#include <tf2_eigen/tf2_eigen.h>
         ^~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
gmake[2]: *** [CMakeFiles/pcl_recorder_node.dir/build.make:90: CMakeFiles/pcl_recorder_node.dir/src/mainROS2.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
In file included from /home/***/ros-bridge/pcl_recorder/src/PclRecorderROS2.cpp:13:
/home/***/ros-bridge/pcl_recorder/include/PclRecorderROS2.h:13:10: fatal error: 'tf2_eigen/tf2_eigen.h' file not found
#include <tf2_eigen/tf2_eigen.h>
         ^~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
gmake[2]: *** [CMakeFiles/pcl_recorder_node.dir/build.make:76: CMakeFiles/pcl_recorder_node.dir/src/PclRecorderROS2.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/pcl_recorder_node.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2

```

## analysis

I found the following cmakelist settings should be wrong:
https://github.com/carla-simulator/ros-bridge/blob/e9063d97ff5a724f76adbb1b852dc71da1dcfeec/pcl_recorder/CMakeLists.txt#L68-L69

there seems  the lack of dependence `tf2_eigen` . 

## suggestion
I changed it as followings, and everything is ok now.
```cmake
  ament_target_dependencies(${PROJECT_NAME}_node rclcpp sensor_msgs
                            pcl_conversions tf2 tf2_ros tf2_eigen)
```

Shall I submit a PR to fix it ?

GoesM, 2024-10-09:
@richie-live I'm not very sure.  I think Jazzy is the latest version so it may face to many confused issues ?  

I am using ROS-humble, Ubuntu 22.04.

richie-live, 2024-10-24:
> @richie-live I solved the problem by the following:
> 
> 1. rm -rf install log build
> 2. sudo apt install ros-humble-tf2-eigen ros-humble-tf2-ros ros-humble-tf2-geometry-msgs
> 3. nano ~/carla-ros-bridge/src/ros-bridge/pcl_recorder/CMakeLists.txt
>    Add the following lines near the top of the file, after the find_package calls:
> 
> ```
> find_package(tf2_eigen REQUIRED)
> include_directories(${tf2_eigen_INCLUDE_DIRS})
> ```
> 
> 4. colcon build --symlink-install
>    please use correct python, ros, carla version
>    Mine is python 3.10, ROS humble (Ubuntu 22.04), Carla 0.9.15
>    Hope it helps

@taeseongkr Thanks! It worked, however I have the following warning:
```
CMake Warning (dev) at CMakeLists.txt:64 (find_package):
  Policy CMP0074 is not set: find_package uses <PackageName>_ROOT variables.
  Run "cmake --help-policy CMP0074" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  CMake variable PCL_ROOT is set to:

    /usr

  For compatibility, CMake is ignoring the variable.
This warning is for project developers.  Use -Wno-dev to suppress it.

```