carla-simulator/ros-bridge #738: ros-bridge couldn't fit into the latest CARLA

GoesM, 2024-07-12:
## description
I'm using the Carla 0.9.15 in ROS2-humble in Ubuntu22.04.

When I launch  the ros-bridge, I faced to the following logs and ros-bridge didn't work.
```sh
[bridge-1] [INFO] [1720767764.270699081] [carla_ros_bridge]: Trying to connect to localhost:2000
[bridge-1] [FATAL] [1720767764.271824643] [carla_ros_bridge]: CARLA python module version 0.9.13 required. Found: 0.9.15
[bridge-1] [INFO] [1720767764.272013648] [carla_ros_bridge]: Shutting down...
[bridge-1] Traceback (most recent call last):
[bridge-1]   File "/home/goes/ROS_Workstation/carla/carla-ros-bridge/build/carla_ros_bridge/src/carla_ros_bridge/bridge.py", line 419, in main
[bridge-1]     sys.exit(1)
[bridge-1] SystemExit: 1
[bridge-1] 
[bridge-1] During handling of the above exception, another exception occurred:
[bridge-1] 
```

I notice the `FETAL` log is 
`[bridge-1] [FATAL] [1720767764.271824643] [carla_ros_bridge]: CARLA python module version 0.9.13 required. Found: 0.9.15`

Does it mean that we stiil can't use the ros-bridge with the latest Carla ?

oliver-schumann, 2024-08-21:
I think you only have to modify the following file https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/src/carla_ros_bridge/CARLA_VERSION

There are several forks of the bridge that work nicely with Carla 15 and newer ROS versions. You might find some help in these repos.

GoesM, 2024-09-05:
it works!  greatly thanks for your help  ^_^

Hynh4u, 2024-11-13:
@GoesM Hello, I also do a project that involve in carla ros bridge. The version we use that like you (Ubuntu 22.04, Carla 0.9.15, Humble). However, the carla ros bridge is not working. Can you give me the instruction in order to use it. Thank you

GoesM, 2024-11-14:
@Hynh4u  what about the issue you've met ?

I just modify the following file https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/src/carla_ros_bridge/CARLA_VERSION with following context:
```
0.9.15
```

and it works.

oliver-schumann, 2024-11-16:
@Hynh4u at first check with nvtop (if you are on Ubuntu) on which card Carla is actually running. 
If you run the simulator from docker (which is the recommended way) check the following doc:

https://carla.readthedocs.io/en/0.9.8/build_docker/

In docker run you can specify with --gpus the visible gpus.

Hynh4u, 2024-11-18:
Hi @oliver-schumann, when i check with nvtop, its show: No GPU is monitor. I run CarlaUE4 with the 'make launch' command and when i find the GPU card in terminal, i found the line: LogVulkanRHI: Display: Device 1: Intel(R) UHD Graphics 630 (CFL GT2)
[image]
and i want to run my CarlaUE4 with this AMD card 
[image]