carla-simulator/ros-bridge #748: [bridge-1] ModuleNotFoundError: No module named 'carla'

ChosenOne-xx, 2024-11-22:
First of all, thank you for this useful open-source project. But when I run “ros2 launch carla_ros_bridge carla_ros_bridge.launch.py“, I faced the error as following:

```
> [INFO] [launch]: All log files can be found below /home/nrist/.ros/log/2024-11-22-10-34-44-946431-nrist-37348
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [bridge-1]: process started with pid [37351]
> [bridge-1] Traceback (most recent call last):
> [bridge-1]   File "/home/nrist/carla-ros-bridge/install/carla_ros_bridge/lib/carla_ros_bridge/bridge", line 11, in <module>
> [bridge-1]     load_entry_point('carla-ros-bridge==0.0.0', 'console_scripts', 'bridge')()
> [bridge-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 490, in load_entry_point
> [bridge-1]     return get_distribution(dist).load_entry_point(group, name)
> [bridge-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2854, in load_entry_point
> [bridge-1]     return ep.load()
> [bridge-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2445, in load
> [bridge-1]     return self.resolve()
> [bridge-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2451, in resolve
> [bridge-1]     module = __import__(self.module_name, fromlist=['__name__'], level=0)
> [bridge-1]   File "/home/nrist/carla-ros-bridge/install/carla_ros_bridge/lib/python3.8/site-packages/carla_ros_bridge/bridge.py", line 24, in <module>
> [bridge-1]     import carla
> [bridge-1] ModuleNotFoundError: No module named 'carla'
> [ERROR] [bridge-1]: process has died [pid 37351, exit code 1, cmd '/home/nrist/carla-ros-bridge/install/carla_ros_bridge/lib/carla_ros_bridge/bridge --ros-args -r __node:=carla_ros_bridge --params-file /tmp/launch_params_kk16_tb7 --params-file /tmp/launch_params_9p0xrj_c --params-file /tmp/launch_params_5ds1kysf --params-file /tmp/launch_params_bb0pzmn3 --params-file /tmp/launch_params_b62i7x27 --params-file /tmp/launch_params_im2f58qx --params-file /tmp/launch_params_duxsu0r6 --params-file /tmp/launch_params_lwujl2xn --params-file /tmp/launch_params_a22i_e1q --params-file /tmp/launch_params_boehkfi9 --params-file /tmp/launch_params_7wehlamj'].
[INFO] [launch]: process[bridge-1] was required: shutting down launched system
```

I intuitively think it's a problem with the version of Carla Egg. There's only "carla-0.9.13-cp37-cp37m-manylinux_2_27_x86_64.whl" in the dir "/carla/PythonAPI/carla/dist", so I download the "carla-0.9.13-cp38-cp38-manylinux_2_27_x86_64.whl " from Pypi.org and pip it. But the problem has not been solved. 

Is there anything wrong with me? How should I solve the error? Thanks again.

environment: carla0.9.13 + py3.8 + ROS2 foxy + Ubuntu20.04

Billthekidz, 2024-12-19:
Don't you have to use python3.7 for carla python package?

ChosenOne-xx, 2024-12-30:
> Don't you have to use python3.7 for carla python package?

No, my python version is 3.8.20. And I have solved the problem . Thanks for your reply.

aSquaredRush, 2025-02-14:
> > Don't you have to use python3.7 for carla python package?
> 
> No, my python version is 3.8.20. And I have solved the problem . Thanks for your reply.

What is the solution? I have the same issue

Billthekidz, 2025-02-15:
guys, you have to use the python version said on the CALRA python package. Otherwise, using a wrong python version, it wouldn't be able to identify CARLA python package as a valid python package for your python version.
For example, if CARLA 0.9.13 asks python 3.7 and you use python 3.8, your python will not identify the CARLA python package. You python version has to be 3.7.

Billthekidz, 2025-02-15:
> > Don't you have to use python3.7 for carla python package?
> 
> No, my python version is 3.8.20. And I have solved the problem . Thanks for your reply.

And hey since many people ask for your solution, I can give a theory: solve it by changing "cp37" in the egg file to "cp38" and python 3.8 will find it. This doesn't mean it will work seamlessly but it gets the first error.